wled/WLED #169: Compile Error

Mazdaracer121, 2019-05-21:
Hey there. Great Project. i used the .bin file for my other Wemos D1 and works fine.

But flashing with arduino ide gives this error output. how to fix that?

Arduino: 1.8.7 (Windows 7), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (1M SPIFFS), v2 Higher Bandwidth, Disabled, None, Only Sketch, 921600"

In file included from C:\Users\Büro\Documents\Arduino\libraries\FastLED/led_sysdefs.h:27:0,

                 from C:\Users\Büro\Documents\Arduino\libraries\FastLED/FastLED.h:41,

                 from sketch\WS2812FX.h:33,

                 from sketch\WS2812FX.cpp:27:

C:\Users\Büro\Documents\Arduino\libraries\FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: conflicting declaration 'typedef uint8_t boolean'

 typedef uint8_t boolean;

                 ^

In file included from C:\Users\Büro\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:28:0,

                 from C:\Users\Büro\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBrightnessBus.h:29,

                 from sketch\NpbWrapper.h:52,

                 from sketch\WS2812FX.h:30,

                 from sketch\WS2812FX.cpp:27:

C:\Users\Büro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/Arduino.h:191:14: error: 'boolean' has a previous declaration as 'typedef bool boolean'

 typedef bool boolean;

              ^

In file included from C:\Users\Büro\Documents\Arduino\libraries\FastLED/led_sysdefs.h:27:0,

                 from C:\Users\Büro\Documents\Arduino\libraries\FastLED/FastLED.h:41,

                 from sketch\WS2812FX.h:33,

                 from sketch\WS2812FX_fcn.cpp:27:

C:\Users\Büro\Documents\Arduino\libraries\FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: conflicting declaration 'typedef uint8_t boolean'

 typedef uint8_t boolean;

                 ^

In file included from C:\Users\Büro\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:28:0,

                 from C:\Users\Büro\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBrightnessBus.h:29,

                 from sketch\NpbWrapper.h:52,

                 from sketch\WS2812FX.h:30,

                 from sketch\WS2812FX_fcn.cpp:27:

C:\Users\Büro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/Arduino.h:191:14: error: 'boolean' has a previous declaration as 'typedef bool boolean'

 typedef bool boolean;

              ^

exit status 1
In file included from C:\Users\Büro\Documents\Arduino\libraries\FastLED/led_sysdefs.h:27:0,

                 from C:\Users\Büro\Documents\Arduino\libraries\FastLED/FastLED.h:41,

                 from sketch\WS2812FX.h:33,

                 from C:\Users\BÃ¼ro\Downloads\WLED-master\WLED-master\wled00\wled00.ino:77:

C:\Users\Büro\Documents\Arduino\libraries\FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: conflicting declaration 'typedef uint8_t boolean'

 typedef uint8_t boolean;

                 ^

In file included from C:\Users\BÃ¼ro\Downloads\WLED-master\WLED-master\wled00\wled00.ino:37:0:

C:\Users\Büro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/Arduino.h:191:14: error: 'boolean' has a previous declaration as 'typedef bool boolean'

 typedef bool boolean;

              ^

Fehler beim Kompilieren für das Board LOLIN(WEMOS) D1 R2 & mini.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

Aircoookie, 2019-05-21:
What a strange error message... Do you have NeoPixelBus version 2.4.3 installed and FastLED 3.2.0?

Mazdaracer121, 2019-05-22:
Yes. i installed the versions you recommend.. 

Same Error Messages...

Aircoookie, 2019-05-23:
Poked around for a bit. Seems like you are running a very recent version of Arduino core which has a compatibility problem with FastLED. You can either downgrade or try the fix provided here: https://github.com/FastLED/FastLED/issues/733

Aircoookie, 2019-10-25:
No longer an issue in recent versions of WLED :)

closed 2019-10-25 by Aircoookie