wled/WLED #170: WebUI is ignored after a while

mentaluproar, 2019-05-21:
After a few days, WLED ignores input from the webUI and other services. So if I try to turn the brightness to zero before bed, it will stay lit and animated anyway. Tried in multiple browsers and multiple devices. 

Stranger still, turning it off, waiting a minute, and turning it back on puts the ESP in a weird state where it won’t load WLEDs interface at all. To reset it, I have to shut it off and wait overnight before giving it power again.

Aircoookie, 2019-05-21:
Seems to me like the ESP loses connectivity to your WiFi (the browser caches the UI so it seems like it still works). I hwill look into it! 
How far away is the ESP from your WiFi access point? Do you use the binary or compiled from source?

You really have to wait overnight? I've had lock-up cases before, but just pressing the reset button or disconnecting power for a second should reconnect it. If you need to wait, maybe your ESP hardware is faulty (or your router has some kind of child policy set where it doesn't allow the ESP to connect at night)

mentaluproar, 2019-05-21:
It’s about 12 feet from the router. I’ve ruled out caching by using other devices. I compiled from source because I couldn’t get the binary in. I don’t remember why. 

The esp failing wouldn’t surprise me.

Aircoookie, 2019-05-23:
With ESP crashes, logs help. But in this case, the software continues to run, but loses connectivity. This sort of thing is very hard to debug. You can try though to open `[ESP-IP]/json/info`, it contains useful info like free memory and esp core version. I suspect that this issue is one of three things, either a hardware issue (you could test whether exactly the same source leads to the same problem on another ESP), the ESP running out of heap memory (in some circumstances it could still be able to serve the UI webpage, but not receive any commands), or the WiFi connection dropping for some reason. Even then, it should auto-reconnect theoretically...

mentaluproar, 2019-05-24:
Twice today it has reset itself to solid orange.  Still accessible and responsive though, so its very odd.  I was able to grab that info for you.

{"ver":"0.8.4","vid":1903252,"leds":{"count":97,"rgbw":false,"pin":[2],"pwr":0,"maxpwr":65000,"maxseg":1},"name":"Covfefe Table lighting","udpport":21324,"live":false,"fxcount":80,"palcount":47,"arch":"esp8266","core":"2_4_2","freeheap":18072,"uptime":2441,"opt":119,"brand":"WLED","product":"DIY light","btype":"bin","mac":"ecfabc053055"}

Aircoookie, 2020-01-07:
Closing since old version/issue :)

closed 2020-01-07 by Aircoookie