wled/WLED #172: MQTT Discovery on Home Assistant Not Working

xxMonkeyxx, 2019-05-24:
First of all, WLED is absolutely amazing!!  It is the perfect solution to my never ending quest for LED control (Donation incoming!!! :) ).  

My issue is similar to the last post in #144 in that HA Discovery is not working.  I configured the MQTT settings and I see the mqttDeviceTopics being sent to the broker, but I never see the HA Discovery topic (or messages).  I am using the precompiled WLED binaries (0.8.4) on WeMos LOLIN D32 (ESP32) and have also tried it on a DOIT ESP32 DEVKIT (NodeMCU like Dev Board) with HASSio v93.1.

No other issues that I have found so far.  I have been using the native UDP sync option between multiple ESP32s and loving it!

xxMonkeyxx, 2019-05-24:
Totally missed this in the release notes for 0.8.4:

**Added HomeAssistant auto discover (broken, please use master. Expect new binaries soon)**

Sooo, nevermind!  I will pull from master or wait for new binaries.

closed 2019-05-24 by xxMonkeyxx

Aircoookie, 2019-05-24:
Sorry for not yet providing new binaries!

Here you go :)
ESP32
https://drive.google.com/open?id=1Zvceusu3KFf8Er1we796WqFCF8Y8r1ks

ESP8266
https://drive.google.com/open?id=1VhJgpKDyj7h5o8BwodmFdtWmLuJXqlUb

P.S. thanks so much for your donation, I'm very happy that you like WLED!

zewelor, 2019-05-28:
@Aircoookie Whats changed in those binaries ? Some compile options ?

Aircoookie, 2019-05-28:
@zewelor There are no new features. The difference is between 0.8.4 release and commit 4b5a3bd3d5fbf108d01ad26665f2e10e521b790e (latest master)
The only changes are that HA autodiscovery works, MQTT will reconnect automatically and a crash with Meteor FX was fixed.