wled/WLED #173: 2 x WLED kommen sich in die Quere.

SeriousOri, 2019-05-25:
Hallo Aircoookie
Ich habe auf 2 Wemos D1 Mini je ein Wled drauf. Einmal eine Kugel und einmal ein LED Streifen unter einem Schreibtisch. Habe beides Wemo' s in der FritzBox Feste Ip's vergeben. Beide Wemo's habe ich anders benannt. Wenn ich Alexa schalte Kugel sage, schalten sich beide ein. Wenn ich Farbwechsle das selbe. Was kann ich machen, das ich die unterschiedlich bedienen kann? 
Wie ist das überhaupt möglich das ich sie gleichzeitig bedienen kann? Trotz Unterschiedliche IP und Name.

Aircoookie, 2019-05-25:
Hi,
das ist das WLED Sync Feature, mit dem sich mehrere Lampen gruppieren lassen.
Einfach in die Sync Einstellungen gehen und den Haken bei "Send notifications on direct change" wegmachen. Dann rebooten und es sollte klappen :)

SeriousOri, 2019-05-25:
Danke dir werde es mal so einstellen.

closed 2019-07-11 by SeriousOri