wled/WLED #174: e1.31 Multicast not working

glitchqc, 2019-05-26:
Hi 
i recently tried your code with e1,31

i manage to make it work with Unicasst (specify the ip address in the software
but when i switch my software to multicast (which is the mode i need) the led stop working and get back to powerup colors.

Aircoookie, 2019-05-26:
Hi!
It is required to reboot WLED once after switching to multicast. Did you try that?

glitchqc, 2019-05-31:
Yes i tried multiple times

sjude68, 2019-08-18:
How do I configure hyperion with wled. Can I use hyperion grabber on Android tv. I have used hyperion installed on Raspberry Pi which worked great.

closed 2020-01-07 by Aircoookie

nethoncho, 2021-12-01:
I am having the same issue with multicast not working.  unicast works even when the config is in multicast. May be a config issue because wled only binds to one or the other based on the config but not both. [See Code](https://github.com/Aircoookie/WLED/blob/17c20276a9641c31ff7e578e1d54766b79d0942d/wled00/src/dependencies/e131/ESPAsyncE131.cpp#L44)
I am using [sACNView](https://sacnview.org/) as a test harness. I will get a build environment going to see if I can track down the issue...

nethoncho, 2021-12-01:
Looking through the source I [found this](https://github.com/Aircoookie/WLED/blob/17c20276a9641c31ff7e578e1d54766b79d0942d/wled00/src/dependencies/e131/ESPAsyncE131.cpp#L80). I can not find where n is set.
`for (uint8_t i = 1; i < n; i++) {`
Can someone have a look?