wled/WLED #175: Add sample home assistant configuration

acid2000, 2019-05-27:
Adding a sample home assistant configuration to the wiki would speed things up a bit. I use the following to control all the leds in my kitchen.

~~~~ 
light:
  - platform: mqtt
    name: "Kitchen Floor Lights"
    command_topic: "wled/all"
    brightness_command_topic: "wled/all"
    rgb_command_topic: "wled/all/col"
    rgb_command_template: "{{ '#%02x%02x%02x' | format(blue, green, red)}}"
    effect_command_topic : "wled/all/api"
    effect_list:
    - "FX=0"
    - "FX=1"
    - "FX=2"
    - "FX=3"
    - "FX=4"
    - "FX=5"
    - "FX=6"
    - "FX=7"
    - "FX=8"
    - "FX=9"
    - "FX=10"
    - "FX=11"
    - "FX=12"
    - "FX=13"
    - "FX=14"
    - "FX=15"
    - "FX=16"
    - "FX=17"
    - "FX=18"
    - "FX=19"
    - "FX=20"
    - "FX=21"
    - "FX=22"
    - "FX=23"
    - "FX=24"
    - "FX=25"
    - "FX=26"
    - "FX=27"
    - "FX=28"
    - "FX=29"
    - "FX=30"
    - "FX=31"
    - "FX=32"
    - "FX=33"
    - "FX=34"
    - "FX=35"
    - "FX=36"
    - "FX=37"
    - "FX=38"
    - "FX=39"
    - "FX=40"
~~~~

Aircoookie, 2019-05-28:
Hi! Sorry you had to come up with all this by yourself. There is supposed to be an autodiscovery via MQTT which adds the config, but it was broken with the 0.8.4 binaries. I've released new ones :)

acid2000, 2019-05-28:
I’m using the firmware you linked in another issue as the MQTT on command wasn’t working.

I also have home assistants discovery featured turned off so I would always be in the situation where I need to add this to the config file.

It’s worth documenting it in case people want to use other API commands for example.

Aircoookie, 2019-05-28:
Alright, I will add a HomeAssistant page to the wiki soon :)

acid2000, 2019-05-29:
Before anyone copies this it’s worth saying the RGB conversion is very naive (or wrong)

robeving, 2019-05-30:
Almost obviously the rgb string should be:
`rgb_command_template: "{{ '#%02x%02x%02x' | format(red, green, blue)}}"`

Aircoookie, 2019-05-30:
Updated in [wiki](https://github.com/Aircoookie/WLED/wiki/Home-Automation-systems)!

closed 2020-01-07 by Aircoookie