wled/WLED #176: Color selection wheel blacked out

mentaluproar, 2019-06-05:
Me and my roommates iPhones (an X, 7, and xr) cannot see the color wheel. It’s solid black. I can still select colors with it, but it’s all black so we can’t really see what we are aiming for.

StormPie, 2019-06-05:
Hey! Are you just using Safari? Does it work using a laptop/computer or an Android device?

mentaluproar, 2019-06-05:
Yes, it’s otherwise fine. I just tried it on Firefox for iOS and it’s fine.

mentaluproar, 2019-06-12:
Any news on a fix?

Aircoookie, 2019-06-12:
Sorry for not replying earlier, I currently lack an iOS device for testing this.
Does it also happen on this page? https://iro.js.org/ If it works fine, we might be able to fix by updating the colorpicker library.

mentaluproar, 2019-06-14:
Yes. That one works fine. I can’t really do much resting on an updated version though because of another issue. My 8266 won’t respond when Bluetooth coexistence is enabled on my router. I turn that off and literally all Bluetooth in the house goes haywire.

mentaluproar, 2019-06-18:
Looks like you were right.  V4 is listed as fixing this very issue in the migration guide.
https://iro.js.org/migrating.html

Aircoookie, 2020-01-07:
0.9.0 UI is now using V4 :)

closed 2020-01-07 by Aircoookie