wled/WLED #177: Adding more relay controls

sjude68, 2019-06-05:
I love Wled and used it on NodeMCU and D1 Mini.  Love the sync function to synchronize all the LED Strips together.  

Also tried the Analog Clock function using 60 LED WS2812B Ring.  Works great.  I have seen other codes on the net but this Wled does all in one. 

This is not an issue but a request.

I have used a relay on D6 (GPIO12) on D1 Mini.  This is directly controlled by the WLED ON/OFF function.  Used to toggle OFF lights other lights and start WLED effects.

How can I add code for Emulating 4 more Alexa switches in the wled06_usermod file so that I can use wled for switching ON & OFF other devices also with Alexa or MQTT control.

Aircoookie, 2019-06-12:
Hi! Awesome that you like WLED!

WLED uses the [Espalexa](https://github.com/Aircoookie/Espalexa/) library. You can quite easily extend the `wled12_alexa.ino` (and/or `wled17_mqtt.ino`) files with your own code (like in Espalexa examples). Just increase the `#define ESPALEXA_MAX_DEVICES` in `wled00.ino` to 5. Let me know if you need help!

sjude68, 2019-06-13:
Thanks Aircoookie.  I tried adding 2 devices in wled12_alexa.ino and they
got discovered.  I used your code to add them.

 device2 = new EspalexaDevice("Fountain", toggleFountain,
EspalexaDeviceType::onoff);
espalexa.addDevice(device2);

But in the checking for on or off, I am not getting the correct status.

EspalexaDeviceProperty m = espalexaDevice->getLastChangedProperty();
  if (m == EspalexaDeviceProperty::on)
  {
    Serial.println("toogleFountain Received ON");
  } else {
    Serial.println("toogleFountain Received OFF");
  }

Always prints OFF.

sjude68, 2019-06-15:
I got 4 switches working through relays. Works well with Alexa.

I am not sure how to configure MQTT for the subscription. Where do I add my keywords to look for. Should I add on the 1 section for ON, on, t, T and then use the if statements

sjude68, 2019-12-19:
@Aircoookie the segment control in v0.9.0 is very good.   I used the master branch and added the modified code for 4-relays in usermode, alexa control & mqtt files.  I have another ws2801b wled strip connected round my TV for Hyperion (running on Raspberry Pi).  Hyperion grabber running on Sony Android TV.

I have noticed that when I switch ON relay 2 (1st ESP D1 Mini) by either Alexa or MQTT the LED strip comes on and also the Hyperion strip (2nd ESP D1 Mini) 1st few leds.  Sync is disabled.  Seems like Relay 2 on 1st ESP and 2nd ESP are linked.  This is not happening on the 1st or 3rd relay.  I have Alexa Echo Dot also on the network.  

How do I give different addresses for each Alexa devices on different ESP Nodemcu.

I want to address another issue here or should I move it another case.

On the 4-relay Nodemcu, I created 3 segments.  Also on the TV Hyperion I created 3 segments.  All are running different Effects.  But sometimes the Effects from 2nd segment continues to 1st segment for some time.  Then they run individually.

My LED strip is a GRB strip.  I purchased another IP68 Waterproof RGB Digital LED Pixel Light.  These are LEDs with builtin controller.  But they are RGB configuration.  When I extend my previous stip to these LEDs the color combination works differently.  Is there a way to use RGB combination on another segment ?

closed 2020-04-24 by stale[bot]

RcRaCk2k, 2020-12-14:
@sjude68 will you share your source-code changes for the public? So i do not need to write this by my own.

sjude68, 2020-12-15:
@RcRaCk2k.. Which source do you need. Is it the 4-relays I added.

RcRaCk2k, 2021-01-04:
@sjude68 Yes the 4-relays you have added.

sjude68, 2021-01-04:
@RcRaCk2k I have attached the file for you. 
[WLED-0.9.1 - 4 Relays - NODEMCU.zip](https://github.com/Aircoookie/WLED/files/5765081/WLED-0.9.1.-.4.Relays.-.NODEMCU.zip)

Hope it helps you.

RcRaCk2k, 2021-01-04:
@sjude68 thank you very much.