wled/WLED #179: Pins above 33?

aenertia, 2019-06-09:
Hi I have been using M5StickC with WLED to test them. 

It seems that pins above 33 in the FastLED/NPP overlays are undefined/unhandled so using high pins for say the buttons( 37/39) or ADC chan0 (36) for the LED Driver channels doesn't seem to work. 

I've also hacked on some basic Display support using the M5Display.h library.

Mainly the PIN definitions seem to need a bit of refactoring for high pins on the esp32 with how things get wrapped/built by the programme structure.

Aircoookie, 2019-06-12:
Hi! 
See NPB doc: https://github.com/Makuna/NeoPixelBus/wiki/ESP32-NeoMethods#neoesp32rmt-methods

> NeoEsp32I2s1800KbpsMethod supports any available pin below 32.

This also sems to be the case for all other ESP32 driver methods. Pins 32+ are addressed differently internally and therefore don't work for high refresh rate purposes. Sorry :)

aenertia, 2019-06-12:
Should be using the I2S method (which is pretty new) then pin36 is the
first ADC0 Channel0 - so this information needs correction.

The Button PIN's however shouldn't be affected by this - but it does
explains why I don't see definitions for PIN > 33 in the NPB.

closed 2020-01-07 by Aircoookie