wled/WLED #181: WLED stops responding when wifi router bluetooth coexistence is enabled

mentaluproar, 2019-06-13:
DS4 controllers, bluetooth mice, august door locks, Apple Watch, basically everything that used bluetooth in my home has been shit for a long time now and we finally figured out why. We moved the router toward the center of the home to improve reception and we didn't realize that it was wrecking bluetooth.  (Hell, it was even screwing with my hue bulbs.  This is a really angry router!)

So I turned on bluetooth pre-emptive mode on my router and everything in the house is fixed, except for the ESP.  The ESP8266 running WLED failed to respond to commands.  Turn it off, it works again.  Turn bluetooth pre-emptive mode on again and the ESP gets less and less responsive until it stops responding all together.

Certainly the esp should be aware of how to handle pre-emptive mode but I have been able to kill it every time, without fail, by turning that feature on.

Aircoookie, 2019-06-24:
Hi, thank you for the report and sorry for the late response!
I never heard of bluetooth pre-emptive mode before, and I can't test it since my router doesn't offer it.
Can you try to flash another firmware like Tasmota to the ESP and see if that works?
I suspect that this isn't something caused by WLED, but by WiFi libraries or the ESP hardware. You can also try WLED 0.8.3, it used a different webserver library.

mentaluproar, 2019-06-24:
There are two Bluetooth fixing features my router supports. Preemptive mode means the router tells Bluetooth devices “this channel is mine, stay away.”  Bluetooth devices that understand preemptive mode obey. Those that don’t may use the same channel anyway. “Enable” basically means the router and Bluetooth devices take turns using the same channel. Bluetooth devices don’t always do this properly though. 

I noticed the same problem on another esp project I’ve been working on. It’s not WLED. It’s the esp. 

I’ve tried asking on the Espressif forums and smallbetbuilder but am not getting any responses. I was hoping you had encountered something like this.

closed 2020-04-05 by stale[bot]