wled/WLED #182: Unable to turn off led strip and control the brightness using Alexa.

daksh1305, 2019-06-28:
Hi @Aircoookie ,

WLED is awesome i am enjoying using it..

The issue which i am facing currently is that i am unable to turn on and off and control brightness of the led strip using alexa.I am using 5050 rgb led with MOSFET it is copletely working fine with the colour.Alexa is able to change the colour but not able to turn off the led and even cannot control the brightness.

I just updated the NPBwrpper.h file which was addressed in #58 but alexa cannot toggle the led on off as well as change the brightness.

Any help would be really appriciated

Aircoookie, 2019-07-15:
Hi!

Sorry for the extremely late reply. Somehow my earlier message didn't get sent.
Can you control on, off and brightness via the web UI? If yes, it is an issue with the Alexa integration.
Make sure you are running the latest Github code and try deleting the light in the Alexa app and then re-discovering it.

I hope we'll be able to solve this problem :)

Def3nder, 2019-10-29:
@daksh1305 I had a similar issue and digged into the issue. Use this version of NPBwrpper.h (together with one define statement in WLED00.ino to get this running:

https://gist.github.com/Def3nder/2bd20d76fda6f3157f4ee4eed89cf35c

closed 2020-04-05 by stale[bot]