wled/WLED #183: Adafruit uploading of Temp

sjude68, 2019-07-01:
I have used Wled to check Temperature using a DHT11 sensor.  I want to upload the values to adafruit.io and also control the WLED strip from there.

I added code to add 4-more relay to control AC devices using Alexa and MQTT.  I want to control the relays using adafruit or any web/android app as Alexa (Echo) needs to be powered ON for Alexa control and MQTT needs to be running.  I am using MQTT on a Raspberry Pi 3.

A little help to configure the files.

I used your help Aircoookie to add the Alexa config (**wled12_alexa.ino**) and MQTT (**wled17_mqtt.ino**).  But can I move the config lines to **wled06_usermod.ino**.

sjude68, 2019-07-04:
I got Temperture  uploading to Adafruit and 4 relays working.

Mariu86, 2019-07-04:
You have all 4 relays connected to 1 nodemcu ?

sjude68, 2019-07-05:
Yes

Mariu86, 2019-07-05:
Very good, :D I hope to see good news from you !

sjude68, 2020-03-09:
@Aircoookie I want to use a DHT11 temp sensor and upload it to adafruit.io

I have seen the example code which uses the code in setup.
`  
// connect to io.adafruit.com
  Serial.print("Connecting to Adafruit IO");
  io.connect();

// wait for a connection
  while(io.status() < AIO_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  // we are connected
  Serial.println();
  Serial.println(io.statusText());
`
I used the code in usermod.ino  The while loop is only to check if AIO_CONNECTED so I omitted it because I cannot use the delay function.

Aircoookie, 2020-03-09:
@sjude68 is it working? That check is probably needed for aio to have time to connect. I'd have an idea, can you send your usermod.ino?

sjude68, 2020-03-10:
@Aircoookie. Its not uploading the temperature to adafruit.io dashboard.  I have attached the usermod.ino file for your reference.  Please ignore the commented lines as I was testing.  

[wled06_usermod.zip](https://github.com/Aircoookie/WLED/files/4310245/wled06_usermod.zip)

Aircoookie, 2020-03-10:
Ok, the connection should be done in `userConnect()`, not `userSetup()`, since at the point userSetup is called, WiFi is not yet connected. Here, I modified your code a bit (not sure if it helps)

[wled06_usermod_new.zip](https://github.com/Aircoookie/WLED/files/4312227/wled06_usermod_new.zip)

sjude68, 2020-03-10:
@Aircoookie it doesn't connect to Adafruit after connection to WiFi.  The debug log as below.

-----------
---DEBUG INFO---
Runtime: 462158
Unix time: 462
Free heap: 11800
Wifi state: 6
State time: 10000
NTP last sync: 999000000
Client IP: (IP unset)
Loops/sec: 4193

Connected! IP address: 192.168.1.170
Init STA interfaces
mDNS started
Connecting to Adafruit IODisconnected!
Access point disabled.
Connecting to Zingat...
---DEBUG INFO---
Runtime: 472162
Unix time: 472
Free heap: 12240
Wifi state: 6
State time: 10000
NTP last sync: 999000000
Client IP: (IP unset)
Loops/sec: 6607

Connected! IP address: 192.168.1.170
Init STA interfaces
mDNS started
Connecting to Adafruit IODisconnected!
Access point disabled.
Connecting to Zingat...
-----------

You are using   WiFi.begin(clientSSID, clientPass);

But in Adafruit it is using AdafruitIO_WiFi io(IO_USERNAME, IO_KEY, clientSSID, clientPass);
and then io.connect()

This is what I have noticed.  I am attaching the complete sketch for your reference.
[wled00.zip](https://github.com/Aircoookie/WLED/files/4313772/wled00.zip)

sjude68, 2020-03-13:
@Aircoookie . I created a new sketch from the usermod.ino for testing and found the upload to adafruit.io working.  But the same in wled is not working.  Can you please look into to check where I am going wrong in the mqtt connection.
[Test_WLED.zip](https://github.com/Aircoookie/WLED/files/4330904/Test_WLED.zip)

Instead of using previous connection method, I used another option found by searching adafruit.

sjude68, 2020-04-10:
@Aircoookie I need a little help from you regarding mqtt->publish

Mqtt publish works when using Raspberry Pi as a Mqtt server.  I found a way to send feeds from Raspberry Pi to Adafruit automatically.  I tested this using MQTT Explorer on Windows 10 by sending mqtt topic manually to Raspberry Pi and it send immediately to Adafruit.io

Help me to configure in wled06_usermod.ino 

I used the line
mqtt->publish("temp", 0, true, celsius); 
but this gives exception(28) error.

closed 2020-08-15 by stale[bot]