wled/WLED #184: Possibility to control several relays

autonomobil, 2019-07-03:
Is there a possibility to control a 4-channel relay module, for example with D5-D8?
[image]

If yes where I start to implement it?

bogeuh, 2019-07-03:
pick a method of control
control the relay with chosen method
then you will know what to add where
work bottom up instead of top down

sjude68, 2019-07-04:
I have used 4 relays with Alexa control.  Add the code as per the example in espalexa.

Bobinox, 2019-11-09:
@sjude68 can you share your code whit us? to take it as example 
thanks in advance

sjude68, 2019-11-11:
@Bobinox 
[wled00.zip](https://github.com/Aircoookie/WLED/files/3829701/wled00.zip)

I added 4 relays and a temp/humidity sensor.  You can check the modified code in attached file.

szurr, 2019-11-13:
Anyone try this with mqtt?

sjude68, 2019-11-13:
> Anyone try this with mqtt?

Yes. I setup MQTT on Raspberry Pi and it works

szurr, 2019-11-13:
Thank you I'll try this

Bobinox, 2019-11-13:
thanks @sjude68 , whit your example i can make it work!
modified some of the code and voilà now i can control my relays whit alexa

sjude68, 2019-11-14:
> thanks @sjude68 , whit your example i can make it work!
> modified some of the code and voilà now i can control my relays whit alexa

Even I modified for Alexa & mqtt control. It was modified within other scripts. I want to move the lines in usermod so when there is a version update, it will be easy to just copy the usermod.ino

closed 2020-04-05 by stale[bot]