wled/WLED #185: Nightlight does not work in combination with several commands

john92648, 2019-07-12:
First of all, this is the best Led program I have seen and installed ever....................
I have a question regarding combination of more commands within a macro.
Here is my example:
win&NL=5&T=1&CY=1&P1=1&P2=10&PT=20000.
The macro works properly but only the Nightlight function will not run. Maybe it can not work because the combination is wrong. I have tried it on several positions within the macro but without success.
On the output XML screen I get this return.
`<vs>
<ac>43</ac>
<cl>255</cl>
<cl>159</cl>
<cl>0</cl>
<cs>0</cs>
<cs>0</cs>
<cs>0</cs>
<ns>1</ns>
<nr>0</nr>
<nl>0</nl>
<nf>1</nf>
<nd>5</nd>
<nt>0</nt>
<fx>9</fx>
<sx>75</sx>
<ix>128</ix>
<fp>0</fp>
<wv>-1</wv>
<ws>0</ws>
<md>0</md>
<cy>1</cy>
<ds>WLED Light</ds>
</vs>`
Any help would be appreciated.
Regards and my congratulation for this awesome project.

Hans Gruber

Aircoookie, 2019-07-15:
Hi, thank you for the kind words and the report!
If you toggle the light with the `&T=1` command, it will always disable the nightlight.
The toggle command is used by the web UI power button and will disable the nightlight (for the common case where the user sets nightlight but turns the light off manually afterwards).

But I believe your use case should be accepted by the system since the toggle and nightlight commands are in the same API request, so I will update the code to work that way.
In the meantime, please use `&A=43` instead of `&T=1`, that should work.

Thank you and I wish you a lot more fun with WLED :)

john92648, 2019-07-15:
Hi, many thanks for your great and friendly support. I changed like you recommend and it works perfect.
greetings from bavaria Germany...........

closed 2019-07-18 by john92648