wled/WLED #188: Wont save wifi settings?

GsynthChris, 2019-07-16:
I'm so bummed out and confused.  I just can't get the super awesome fantastic wonderful software to work.....
I have tried 4xD1minis and a NodeMCU - all esp8266
I tried compiling in Arduino IDE. It compiles and flashes and I can connect to the AP and check out all the effects and such but when I try to go to settings to set up the wifi and such it crashes.  I was able to enter my wifi info in the sketch and it would connect to my wifi then but I still get stopped at the settings menu. However, for the life of me I cant find where to enter my mqtt broker's name and password.  Makes me wish I had never set up a password for my mqtt broker but I would have to change settings on 50+ devices to change that now lol.  Oh, and when I enter my LED count in the sketch it still starts up with only 30.

I tried flashing the pre-compiled binaries (0.8.4 and 0.8.3 ) and I can get to the settings menu and can even enter my LED count and that will save but when I hit save and reboot in WiFi the esp does reboot (or crashes) and immediately starts back up in AP mode (ping doesn't even skip a beat) despite asking it to wait 120 seconds before broadcasting the AP.

I have also tried connecting to it from my phone, tablet, and computer and it still behaves the same.  I have no idea what I could possibly be doing wrong.  I haven't found a similar issue in the other issues. 

I'm sorry to complain.  I REALLY love what you have done here.  I cant wait to get this figured out so I can use this on all my LED strips.  The sketch I have been using is SOOOO much less feature rich than this one. I can't believe how much you have squeezed into this and it just barely fills half the storage in my D1mini! lol

Thanks for you time

Aircoookie, 2019-07-16:
Hi!
I'm happy that you like WLED and thank you for the report!
I'm sorry that there seem to be issues with getting the connection to work.
It could be that the connection to your home network actually works but AP mode is still on (the ESP can do both at the same time). You can delete the "WLED-AP" from the AP Name textfield and it will only start up the AP if it fails to connect to your home wifi. If you still have the AP open, you can try to go to WiFi settings after more than half a minute after booting and you should see Client IP display the IP of the module in your home network.
Also check that the WiFi signal is strong enough, is 2.4Ghz and you didn't make a typo.
If it still doesn't work, I will give you further troubleshooting help!

About MQTT credentials, they are currently not supported because of security concerns. But since many users have requested being able to use them, I will add them in the next version :)

Greetings and I hope you'll be able to have a lot of fun with WLED soon!

sansillusion, 2019-07-16:
I have had the exact same issue. Turns out that my dd-wrt router is the
issue. None of my esp32 will stay connected on it. And my esp8266 has a lot
of issues. But if I run an AP(sharing it's internet connection) on my
laptop it connects fine and I can go in settings.

I hope that a fix for my router comes out sometime soon since I dont like
having my laptop run all the time just to create an access point.

PS: not only wled but all codes using wifi have issues on my router. Some
more basic will last longer before a failure but it always end up crashing.
On my computer AP no crash at all.

GsynthChris, 2019-07-21:
man, I just don't know what I am doing wrong.  How well do esp8266 chips handle this sketch? Does it push the limits? I just doing understand why I am having such a hard time. I seem to be the only one struggling lol.  I tried your suggestions, still no improvement.

timothybrown, 2019-08-13:
> man, I just don't know what I am doing wrong. How well do esp8266 chips handle this sketch? Does it push the limits? I just doing understand why I am having such a hard time. I seem to be the only one struggling lol. I tried your suggestions, still no improvement.

I'm getting the same issue. If I compile from the GitHub master branch (with PIO) the WebUI crashes immediately when I try to change any settings. If I flash one of the pre-compiled binaries I can get into the settings, but it won't save. It just reboots and comes back up into AP mode.

I know my router is *not* the issue, as I have 7 other ESP8266 devices on my WiFi that work fine. (6 of them are smart bulbs running Espurna, the other is custom IR->MQTT software running on a NodeMCUv2 board).

I'll continue to debug and see if I can find the issue.

timothybrown, 2019-08-13:
Well, I got it working! In `platformio.ini` I changed `platform = ${common:esp8266.arduino_core_2_4_2}` to `platform = ${common:esp8266.arduino_core_2_5_0}` under the `[common:esp8266]` section. This seems to work both with the v0.8.4 tag and master branch.

GsynthChris, 2019-09-09:
By the way, I think My issue may have been a weak power supply.  I tried it again tonight with a stronger power source and its working perfectly!

closed 2019-09-09 by GsynthChris