wled/WLED #190: Useing basic segment support

Physikus74, 2019-07-18:
hello and many thanks for this very comfortable and powerful software. I use it in a configuration with about 950 WLEDs. Via the web interface I can adjust colors and select effects. This works great!
I would now like to use the announced "basic segment support". I would like to display other colors or other effects on some of the LED's. In the webinterface I can't find any possibility for the settings. What do I perhaps do wrong, or did I misunderstand the function "basic segment support"?
Many greetings,
Physikus74

Aircoookie, 2019-07-23:
Hi!
Welcome to WLED, I'm glad you like it!
Unfortunately, I'm still working on the segment feature, it is not yet controllable via the UI, only via the [JSON API](https://github.com/Aircoookie/WLED/wiki/JSON-API).

I'll let you know once the feature is ready!
Many greetings and have a nice week!

Physikus74, 2019-07-29:
Hi,
thank you for the answer. I will have to try out the possibilities of the JSON API... ;o)
In which period of time is the configuration via UI to be expected.

Many greetings!

Aircoookie, 2019-07-30:
Awesome!
I believe it will be no more than 1-2 months until I release a big update including segment support in the UI. I'll let you know!

Physikus74, 2019-07-31:
...that sounds great! I'm really looking forward to it...

Many greetings,
Physikus

brentbrooks70, 2019-08-21:
@Aircoookie looking at playing with segment support via JSON API, in your [Wiki](https://github.com/Aircoookie/WLED/wiki/JSON-API) example it show a start:0 
Stop:20
Len:20

Wouldn’t the length be 21 or or the stop should be 19 for a Len of 20? Am I missing something?
Sorry to be a pain, but again, great project!!! Thanks for all your effort in this.

brentbrooks70, 2019-08-25:
> stop | 0 to info.leds.count | LED the segment stops at, not included in range. If stop is set to a lower or equal value than start (setting to 0 is recommended), the segment is invalidated and deleted.
len | 0 to info.leds.count | Length of the segment (stop - start). stop has preference, so if it is included, len is ignored.

Never mind, just had to RTFM!!

CaffeinatedFunctionality, 2019-10-21:
I am so happy you are doing this. You also made me really happy with the Halloween eyes setting. That is amazing, because I was just about to move to custom software because it didn't have that. If you need any help with this let me know. I would love to help development somehow or add new effects. I mostly work in C# and JS at the moment, but working on understanding your code as well. It's written quite well and seems pretty readable. Thanks for your work on this.

Aircoookie, 2019-10-25:
@CaffeinatedFunctionality thank you for the kind words :) New UI with segment support will be coming in 0.9.0 very soon :)

ndtfl, 2019-10-26:
Hi, thanks for the great project, I have enjoyed this a lot.  Now about the segment support, I m sure many are anxiously waiting... in the meantime, I have 2 questions after reading the wiki
1 - wouldnt it make sense to allow the "stop" value bigger or smaller than start as long as they both inside the range (between 0 and led count - 1), the reason being when I have start bigger than stop, I would have the effect in reverse? For example, segment 1: 0-49 and segment 2: 50-99 in 2 lines then when I have a chase effect, both will run in the same direction, but if I have the second one as 99-50, then the 2 segments as 2 lines running in opposite direction, this will be much nicer?

2 - If I m not mistaken, currently only 1200 leds are supported, would you consider increasing this as segments will be more useful if longer strips are used and therefore more leds counts?

Many thanks for your time in this project and I wish you well.  Thanks

brentbrooks70, 2019-11-08:
> 1 - wouldn't it make sense to allow the "stop" value bigger or smaller than start as long as they both inside the range (between 0 and led count - 1), the reason being when I have start bigger than [...]

@ndtfl You could just set the **"rev"** flag for that segment to true, having both methods working simultaneously could become confusing.  @Aircoookie ?
> 2 - If I m not mistaken, currently only 1200 leds are supported, would you consider increasing this as segments will be more useful if longer strips are used and therefore more leds counts?

@ndtfl These settings were just increased in [ver 0.86](https://github.com/Aircoookie/WLED/releases) to:
> Increased max. LEDs to 1500 (500 in DMA pin 3 mode)
-> I still recommend staying below 900 LEDs (300 for DMA) for optimal performance

brentbrooks70, 2019-11-08:
@Aircoookie been playing with segments again and have a strange issue. First off, let me say one more time that I absolutely love this project!!! 

I am running 6 strips of 50 LEDs wired in parallel with LEDs and MCU both attached the same 5V 30A power supply. **Here is my json/info (just in case):**

`{"ver":"0.8.6","vid":1910255,"leds":{"count":50,"rgbw":false,"pin":[2],"pwr":829,"maxpwr":4000,"maxseg":10},"name":"Floor Lights","udpport":21324,"live":false,"fxcount":83,"palcount":50,"arch":"esp8266","core":"2_5_2","freeheap":18048,"uptime":798125,"opt":119,"brand":"WLED","product":"DIY light","btype":"bin","mac":"a020a605aaed"}`

If I **POST**:

`{on: true,bri: 127,transition: 0,ps: -1,pl: -1,nl: {on: false,dur: 5,fade: true,tbri: 3},udpn: {send: false,recv: true},seg: [{id: 0,start: 0,stop: 24,len: 25,col:[[255,0,0],[0,255,0],[0,0,255]],fx: 3,sx: 218,ix: 255,pal: 0,sel: false,rev: false,cln: -1},{id: 1,start: 25,stop: 49,len: 25,col:[[0,255,0],[0,0,255],[255,0,0]],fx: 3,sx: 218,ix: 255,pal: 0,sel: true,rev: true,cln: -1}]}`

to _http://IP/json/state_ I get a statusCode: 200 with a payload response of {"success":true} but only segment ID:0 is affected (starts _Wiping_ from red to green and then back to red on the first 25 LEDs as expected) but the segment ID:1 just turns Black (or Off) instead of _Wiping_ from green to red and back to green.

If I **GET** _http://IP/json/state_ it returns:

`{"on":true,"bri":127,"transition":0,"ps":-1,"pl":-1,"nl":{"on":false,"dur":5,"fade":true,"tbri":3},"udpn":{"send":false,"recv":true},"seg":[{"id":0,"start":0,"stop":24,"len":24,"col":[[255,0,0],[0,255,0],[255,255,255]],"fx":3,"sx":218,"ix":255,"pal":0,"sel":false,"rev":false,"cln":-1},{"id":1,"start":25,"stop":49,"len":24,"col":[[0,255,0],[255,0,0],[0,0,0]],"fx":3,"sx":218,"ix":255,"pal":0,"sel":true,"rev":true,"cln":-1}]}`

which tells me it should be working......Since segment ID:1 is the selected seg I can change things in the GUI and the appropriate changes are reflected in the json/state but everything is still Black/Off on that segment. 
So I simplified it and set it to **FX**:0 on both segments so I would expect to see 1st 25 red and 2nd 25 green and the _json/state_ verifies the changes were accepted but segment ID:1 is still off

`{"on":true,"bri":127,"transition":0,"ps":-1,"pl":-1,"nl":{"on":false,"dur":5,"fade":true,"tbri":3},"udpn":{"send":false,"recv":true},"seg":[{"id":0,"start":0,"stop":24,"len":24,"col":[[255,0,0],[0,255,0],[255,255,255]],"fx":0,"sx":218,"ix":255,"pal":0,"sel":false,"rev":false,"cln":-1},{"id":1,"start":25,"stop":49,"len":24,"col":[[0,255,0],[255,0,0],[0,0,0]],"fx":0,"sx":218,"ix":255,"pal":0,"sel":true,"rev":true,"cln":-1}]}}` 

I did notice that both segments that are **len**:25 are reporting **len**:24 (I am assuming this means 0-24 actually giving 25LEDs). I removed the **len** entries from my **POST** but they stay the same. I have also put **len** back in and removed the **stop** entry in my **POST** but get the same state reported back. 

Any thoughts or suggestions?

Aircoookie, 2019-11-08:
@brentbrooks70 Thank you very much for your detailed findings! I was able to replicate the issue and found not one, but two bugs!

Firstly, the `select` attribute is totally broken, because a) the effect config for the first segment is not applied if it is not selected, and b) any subsequent segment's config will be overriden with the first segment's if they are selected. So for everything to work correctly, the first segment (id 0) must be selected and all other segments must be unselected.

Secondly, the `rev` attribute is also broken if the start of the segment is not 0.

Fixes are on their way!

Physikus74, 2019-11-12:
Hello,
that explains some effects I also had. ;o) I'm looking forward to the fixes...

marc-gist, 2019-11-13:
yay! I ran into some strange behaviour as well today, glad your on top if it @Aircoookie ! Thanks for the project!

Aircoookie, 2019-12-28:
Hope you all like the new 0.9.0 version with segment support in the UI :)
Just a little note to self to add custom name support to segments soon!

lastred1, 2020-01-22:
So, inquiring minds would like to know, from @brentbrooks70's example, what should the json look like for it to work?

Aircoookie, 2020-01-22:
@lastred1 In WLED 0.9.0, @brentbrooks70 POST command works as expected for me :)

closed 2020-05-28 by stale[bot]