wled/WLED #194: Adalight not working

filipviduna, 2019-07-28:
Using **Arduino Nano** i can stream from PC to **ws2812b** via **Prismatik** with no problems at all. But when using WLED on **NodeMCU**, the stream is delayed and just not working properly. Is it a **NodeMCU** problem? Should i connect the **Nano** before **NodeMCU** to grab and pass the data? Thanks for any advice.

Aircoookie, 2019-07-30:
Hi! Sorry that there are issues with Adalight!
Can you send me the code that you use with the Arduino Nano?
Maybe the serial baudrate of something is off. I will check if it is a problem with WLED!

No, WLED can't process the raw pixel data coming from Arduino nano. You could use an Arduino relay module and connect it to GPIO12 of NodeMCU, then connect the led data lines from both the Nano and NodeMCU to it. This can automatically switch between the nano ambilight and WLED.

Ideally thought, WLED should work nicely with Adalight. How many WS2812b leds are you using?

ViFi-connected, 2019-07-31:
I am using 116 leds. Baudrate is set to 115200 in Prismatik (works with Nano, same USB port). 

[Adalight_ArduinoNano.txt](https://github.com/Aircoookie/WLED/files/3453014/Adalight_ArduinoNano.txt)

codamo35, 2019-08-04:
Hey Aircoookie, great stuff cooking in here. Thoroughly enjoying it!! I had a similar issue a while back with a Prismatik/Adalight setup, but didn't extensively troubleshoot it. Just spent a bit of time today looking back into it and here's what I'm running into:

-I have 400 addressable points in my ceiling, but tested this down to just 20 LED points (changing length had no effect).
-I have the WLED 0.8.4 binary loaded onto a NodeMCU, which works fine and dandy with controlling all 400 points. I have the Prismatik 5.11.2.19 binary application loaded to my Mac OS 10.14.5, and the application functions fully with no issues. 
-Baudrate set to 115200 in the Prismatik application and I have my WLED device hooked to my computer via USB (with correct COM port set in Prismatik). 
-With Prismatik running WLED sees incoming commands and stops producing it's own colors, and then produces a frozen color projection across all LEDs. If I display a moving color-wheel video on my computer screen under the marked pixel-points, the LEDs change color every 7-45 seconds, although sometimes not at all. It's not predictable and not repetitive. Each instance when it changes it is a stationary color change, no movements, just frozen.
-If movement stops on my computer under the pixel points, WLED switches back to producing it's previous color mode after the timeout of not receiving any new Adalight commands (timeout duration set under WLED menus, and work as it should).
-If I create movement under those pixel points again, WLED switches back over to (sort-of) receiving the Adalight commands.

Seems _almost_ there, but something like a refresh rate or interrupt rate not working quick enough. This post seems to point that it's not isolated to me as well. If you need a visual of what is happening I can work up a video recording of the screen / LEDs. Just let me know.

**Prismatik software, tested on both Mac and Windows (and giving same issue on both OS):**
https://github.com/psieg/Lightpack/releases

Aircoookie, 2019-08-20:
Thanks for your feedback! i'll hopefully find a fix soon!

raptusis, 2019-11-15:
Hi! Any news for this very interesting feature?

Thanks for all

Stephan3, 2019-11-29:
+1 for this. I fiddled most by using the hints in several topics here. As i use Hyperiond i wanted to bind it using the adalight protocol. My ESP32 is on a selfcompiled version, its latest Master.

I saw that hyperion can be used using the UDP protocol. This is lagging every now and then. So next step for me was trying adalight. If i missed a thing in the wiki, explaining how to do it richt, sorry for that.

Aircoookie, 2019-11-29:
@Stephan3 I merged the new Adalight parser yesterday! Are you already using that version? If yes, I need to look into that, losing connection should never happen, especially when data is only sent via USB/serial.

Stephan3, 2019-11-29:
I pulled master yesterday,too. As we can’t be sure. Let me pull latest tomorrow, test and report back.
:)

codamo35, 2019-12-15:
Can confirm, 0.9.0-b1 inclusion of #391 has resolved this original issue!!

closed 2019-12-15 by Aircoookie