wled/WLED #195: wifi captive portal

richyxi, 2019-07-29:
hi, im trying to put this in a bar of my friend, and his network has a captive portal to access, im thinking of how can i connect the esp module into his net, is it possible?

Aircoookie, 2019-07-30:
Hi!
Unfortunately the ESP can't complete the captive portal to gain internet access, but maybe that isn't even needed. In most of the cases the ESP will connect to the router just fine despite a captive portal being configured, and you can still access the ESP from other devices in the network if you know it's IP. If that doesn't work (because the Captive portal may block all traffic, even in the local network), the only way is to assign a static IP to the ESP and add an exception in the router configuration to disable the captive portal for that IP. 
Sorry if there are any issues, but I'll hope you get it working!
An alternative would be to just use the standalone "WLED-AP" accesspoint mode without connecting to your friends network. You can change the password of the ap or hide it's SSID so that bar guests won't mess with the lights.

richyxi, 2019-07-31:
Thank you very much my friend

closed 2020-01-07 by Aircoookie