wled/WLED #196: Alexa color support

foliveira15, 2019-07-29:
Hello
I can control ON/OFF and brightness with Alexa but colors is not working.
There is any special configuration that I have to change?
Thx

brentbrooks70, 2019-07-30:
Are you using precompiled binaries or compiling yourself? If precompiled, which version? If compiling, what IDE version, type of board, board core version are you using? You could also just post the results from http://<DeviceIP>/json/info.

Aircoookie, 2019-07-30:
In addition to brentbrooks70's suggestion, what Echo model are you using? There is a known issue with Echo Dot 1st and 2nd gen where color control might not work in some cases. You can also try deleting the light in the Alexa app and then rediscover it, that can help at times.

foliveira15, 2019-07-30:
> Are you using precompiled binaries or compiling yourself? If precompiled, which version? If compiling, what IDE version, type of board, board core version are you using? You could also just post the results from http://_DeviceIP_/json/info.

I'm compiling myself using VSCode 1.36.1, Plataformio 2.2.0 Core 4.0.0, ESP32 DEVKIT V1
JSON:
`{"ver":"0.8.5-dev","vid":1906201,"leds":{"count":60,"rgbw":false,"pin":[14],"pwr":0,"maxpwr":65000,"maxseg":10},"name":"WLED Light","udpport":21324,"live":false,"fxcount":80,"palcount":47,"arch":"esp32","core":"v3.3-beta1-179-ge931fe9f5-dirty","freeheap":229264,"uptime":98202,"opt":103,"brand":"WLED","product":"DIY light","btype":"dev","mac":"240ac431de3c"}`

Espalexa result:
```
Hello from Espalexa!

Value of device 1 (Light): 255 (Extended color light, colormode=hs, r=255, g=0, b=0, ct=383, hue=0, sat=254, x=0.50, y=0.50)

Free Heap: 235336
Uptime: 98324052

Espalexa library v2.4.3 by Christian Schwinne 2019
```

> In addition to brentbrooks70's suggestion, what Echo model are you using? There is a known issue with Echo Dot 1st and 2nd gen where color control might not work in some cases. You can also try deleting the light in the Alexa app and then rediscover it, that can help at times.

Using Echo Plus 2nd generation

foliveira15, 2019-07-30:
Ohhh wait! I figure out.... If I have any effect applied, the color don't change... Only if a solid color is active.. I don't know if this is the expected behavior or not....

Aircoookie, 2019-07-30:
Yes, there are quite a lot of effects where the currently selected color doesn't matter, like the Palette effects. This is expected. If the color does not change for ANY effect but Solid, i'll have to look into it. You can try the simple Blink effect, for example.

closed 2020-01-07 by Aircoookie