wled/WLED #197: NodeMCU & WS2812b, not lighting up

OliPassey, 2019-07-30:
Hey,
I have a 5m LED strip of 300 WS2812b, powered by a 5v 10A supply - larger psu is coming.
I've managed to get the software side of things up and running, can access the gui on my home network etc.

However i can't get the strip to light up. I have used ws2801 before, but this is my first time on newer pixels so i may have missed something.

I've connected the Data Pin to D4 on my NodeMCU, along with GND, then GND & Power are connected seperately to the PSU.

Sending commands seems to work, but nothing lights up, no flickering, nothing. This is when things are connected upstream of the arrows pointing down the line of LEDs.

I tried connecting the other-end of the strip, and the first led lit up blue - but no commands worked.

I flashed directly from the binary on here, rather than building and compiling via Arduino - do i need to add config anywhere? Or do i need to install Arduino IDE and recompile ?

Any advice would be appreciated, including where i could see some logs.
Thanks for what looks like a great project
Oli

sansillusion, 2019-07-30:
Doesn't wled uses pin D2 ?

OliPassey, 2019-07-30:
I read online that's the pin I needed, maybe it was an old article.
I'll try d2 when I get home.
Thanks.

Aircoookie, 2019-07-30:
Welcome and sorry for the issue!
WLED in fact used pin GPIO2, this often causes confusion because the pin is labeled D4 on NodeMCU. So, the pin you use is correct. 10Amp power supply is also plenty.
Did you set the current limit in LED settings to 10000mA? The default of 800mA might be causing this.

No, the binary is fine, you should have no issues using it out of the box. If the control UI is accessible, the software is running correctly.
You can try connecting the data cable to the second LED, maybe the first one is broken somehow. Another suggesting would be to use a 3.3v->5v levelshifter (connecting to the LEDs directly is technically out of spec because the voltage is too low). Before that, try connecting your PSU to the other end of the strip, the voltage drop may be high enough for the leds to recognize the signal.

Let me know if none of those suggestions work, I have some more ideas!
I hope we'll get WLED working for you!

OliPassey, 2019-07-30:
Thanks for the reply and ideas, i'll give those a try this wkend as will need to order some bits.
Im worried ive got a dead strip - can i do a StrandTest in the Arduino IDE like i could on WS2801 ?

Aircoookie, 2019-07-30:
Yes sure, you should be able to upload any test code in the Arduino IDE. (you will need to reflash WLED afterwards, but the configuration you already did will be saved). Using a StrandTest will be a good idea to see if it is the LED's or WLED's fault.

OliPassey, 2019-08-01:
Do you have a wiring diagram for how the level shifter should be wired in? I see loads of differing diagrams online, most including a resistor. Don't want to break things further by mis-wiring this thing.
[image]

OliPassey, 2019-08-01:
Ran the data line through the level shifter, like this:
[image]
Not sure if its correct, but it still didnt light anything up.
Im just running a blink test via Arduino IDE - seems to load ok, and the LED on the board lights up, which i think means its sending data as its the same port used?

Not used a breadboard or a level shifter before so i may have done something stupid.
Any ideas?
Really appreciate your assistance on this :)

OliPassey, 2019-08-01:
Not the easiest photo to read:

D4 on the Node MCU is connected to the level shifter, then level shifter onto the LED Strip data pin.
GND is connected to GND
Power is connected directly to the strip

Mariu86, 2019-08-02:
You must provide a 3,3v to LV , a 5v to HV the ground must be to both side this is for 2 pins in the middle

OliPassey, 2019-08-02:
Just trying to fully understand what you are saying;
Do i need two power supplies then? Like this:
[image]

Mariu86, 2019-08-02:
Yes

Aircoookie, 2019-08-02:
I'm also not sure on how to wire up your levelshifter... Just that D4 needs to be connected to input and the led to the output of the levelshifter. No, you don't need a second PSU, Nodemcu has a 3.3v pin :)

OliPassey, 2019-08-02:
Ahh great stuff, will give it another go tonight.
Thanks both!

OliPassey, 2019-08-08:
Well, i owe you an apology - i had a ground fault, sorry!
All is working a treat now, thanks for a great project and the assistance you offer.
Cheers!

closed 2019-08-08 by OliPassey