wled/WLED #199: Question about Buttons

jeckyllhavok, 2019-08-05:
Hi there, 
first of all, Thanks for that great piece of code. I tested many WS28xx Codes for ESPs but none of them is so amazing than WLED.
I got on little question regarding Buttons: Is it possible to dim up and down with a button?
For Example: 
Long press -->Dim up until release
Long press again-->Dim down until release

Also is it possible to switch through the effects with a button?
For example:
Press button-->Effect 1
Press button-->Effect 2
and so on

Thanks in advance

Aircoookie, 2019-08-10:
Hi!
I'm glad that you like WLED!
Dimming with the button is not supported, however I believe that it is a great idea, so I will add it soon!

For switching through the effects, go to time settings and change these settings:
[image]

[image]

Then you can step through the effects with a single click and turn on/off with a long press.

Beliar-666, 2019-08-31:
Hey, just tried this, exactly like the pictures shows. But only Longpress works for on/off. Switching trough the FX-Mopdes doesn´t work. On shortpress nothing happens.

brentbrooks70, 2019-08-31:
@Beliar-666 If you use your browser to enter http://_DeviceIP_/&win&M=1 does it cycle through the effects successfully? After the first time you should just be able to refresh the browser to get it to continue cycling the effects. This works for me and assigning it to any button press (short | long | double) seems to work flawlessly, it evens wraps from last to first effect. Would you post the results of http://_DeviceIP_/json/info to aid in diagnostics.

Beliar-666, 2019-08-31:
@brentbrooks70 no, it wont work with http://DeviceIP/&win&M=1 and http://DeviceIP/json/info only says not found. i´ve 083 cause all other versions won´t work for me. i ever get bootloops with 084 or 085.

brentbrooks70, 2019-08-31:
@Beliar-666 Looking through some of the [commits](https://github.com/Aircoookie/WLED/commit/9b7db548a261935a0ca4cfc19f99dad3520dab8a) it looks like there were some changes between 0.8.3 & 0.8.4 having to do with the button and AP102 compatibility. Do you compile your own code or run precompiled binaries? If you are compiling, what IDE and version, what platform (8266 or 32) and its associated core version, the version (VID) of the code (can be found in WLED00.ino in date format)? You could also look in NpbWrapper.h for the BTNPIN definition and make sure it appears correct. I am sure I have forgotten to ask something, hopefully this will either point you in the right direction or when you answer the above info I might be able to help more.

Beliar-666, 2019-08-31:
0.8.4 both tried. the downloadable bin and self compiled. Both have the same error after upload. My LEDs light up and the flashes white mixed wit5h different colours, then light orange, and so on. so it seems the node mcu 8266 reboots all thje time. 0.8.5 is self compiled. after flash i short get into the Frontend and then also reboots. Tested with 3 Nodemcus. The 0.8.3 runs without error, but has no json.

Beliar-666, 2019-09-01:
I got the error i had. The reboots were caused by mqtt. The settings and datapoints in my Homeautomation from wled 0.8.3 caused the reboots. Turned off mqtt in my automation, deleted the datapoints from wled 0.8.3 restarted mqtt and everything is fine.Now i again have the same datapoints without any errors.

Ashok1900, 2019-09-03:
hello all,

Do you have an schemtic to connect a button.

Ashok1900, 2019-09-03:
sorry, i want to connect a button to my nodemcu v3 so i can scroll trough the effect. i tried it with a pull up resistor but it will not work. i think i do somthing wrong and that's why  i'am looking for a schematic. thanks in advance

Mariu86, 2019-09-03:
Tell us what you do and we will see what happening, I have the button working.

Ashok1900, 2019-09-03:
[image]
 I connected the button like this and used an pull up resistor 10k
I  Also  set the macro settings like setting at the top. But it does not working. I think i do something wrong. I hope you can help me.  I like Wled00 very much!!

Ashok1900, 2019-09-04:
[image]

@Mariu86  how did you connect your button. Can you please give me more information about that ?

Mariu86, 2019-09-04:
I will look in my setup today and I will send you a picture , I think I didn't use any resistor the node mcu hase a resistor build in .

Mariu86, 2019-09-04:
@Ashok1900 my push button it's connected to my nodemcu directly to d3 and vin pin.

Ashok1900, 2019-09-05:
I will try it and let you know today.

Ashok1900, 2019-09-05:
@Mariu86  Yes it's working now. I thank you very much for your great support.

neilvangeffen, 2019-09-26:
Is button brightness still likely to be implemented? I want to use wled for some camping lights, and would be great if i could turn the thing on/off and adjust brightness with a single button!

Aircoookie, 2020-03-29:
Button brightness adjustment added to list of things to do :)

Aircoookie, 2020-03-29:
Closing since the brightness on long press was also proposed in #760

closed 2020-03-29 by Aircoookie

Mopfus, 2022-03-26:
WLED is really super cool! I am in the process of moving my system completely to WLED at the moment. My former control of brightness by holding the hardware button has worked well so far (done with ESPhome NeoPixel). Now I wanted to implement the same in WLED. But I fail at it. At least setting it up the way it was before. So far I could only achieve e.g. single press = 10% brighter, double press 20% darker with the macros. But I need these key-presses for other functions.

**My previous control was as follows:**
When holding the button, the brightness changes between 1% and 100% until the button is released. At 1% and 100%, it pauses briefly and blinks (so the user knows this is the maximum). The idea actually came from the control of a headlamp.

I hope you can understand what I mean. Here is a short video which should make it understandable easily:
https://user-images.githubusercontent.com/80747454/160247481-01247e5e-1965-4df2-9e6a-96cf31842a19.mp4

**Is there any way to implement something similar with WLED?**

If not, I think this would be a great feature. I can report from my own experience that this method is really very suitable for controlling the brightness of a room lighting.

Thank you very much for your suggestions, tips or ideas.

_My setup:
D1 mini, 120 WS281B-LED, HomeAssistant with ESPhome (programming this was done with ESPhome NeoPixel)_

PS: In the future I wanted to attach a second button, with which I can then change the color temperature (Kelvin) in the same way. ;-)

neilvangeffen, 2022-03-26:
You can accomplish that with home Assistant, not direct with WLED.
Still hoping we can one day, but doesn't seem important enough to others. I want to run WLED stand alone in a camper, but can't just yet.
This won't give you enough, but this is what I did
[image]
Not near a proper machine to export it for you, screenshot on phone is best I can do atm

neilvangeffen, 2022-03-26:
[{"id":"c2850c8c.6ab43","type":"switch","z":"d7684e86.ed817","name":"Light Full","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"On","vt":"str"},{"t":"eq","v":"Toggle","vt":"str"},{"t":"eq","v":"Ramp","vt":"str"},{"t":"eq","v":"Direction","vt":"str"},{"t":"eq","v":"Release","vt":"str"},{"t":"eq","v":"Off","vt":"str"},{"t":"eq","v":"Hold","vt":"str"}],"checkall":"false","repair":false,"outputs":7,"x":720,"y":760,"wires":[["c7f42c0e.938b9"],["c7f42c0e.938b9"],["a31687ed.649d78"],["251d7643.0d5c4a"],["d61c5629.20a0a8"],[],[]]},{"id":"87db1b79.348458","type":"api-call-service","z":"d7684e86.ed817","name":"Ikea On","server":"3503d7cc.127568","version":3,"debugenabled":false,"service_domain":"light","service":"turn_on","entityId":"light.ikea_left, light.ikea_right","data":"{    \"brightness\": {{payload}},    \"transition\": \"0\"}","dataType":"json","mergecontext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":1840,"y":800,"wires":[[]]},{"id":"8771abf3.cde338","type":"change","z":"d7684e86.ed817","name":"Bright Up","rules":[{"t":"set","p":"payload","pt":"msg","to":"Bright Up","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":1220,"y":760,"wires":[["7b4976d0.d59648"]]},{"id":"ca60bf95.c9be1","type":"change","z":"d7684e86.ed817","name":"Bright Down","rules":[{"t":"set","p":"payload","pt":"msg","to":"Bright Down","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":1230,"y":800,"wires":[["a16a4dcf.d9542"]]},{"id":"ce64f707.1d26b8","type":"switch","z":"d7684e86.ed817","name":"Dim Dir","property":"ikeaDimDir","propertyType":"flow","rules":[{"t":"eq","v":"1","vt":"num"},{"t":"neq","v":"1","vt":"num"}],"checkall":"true","repair":false,"outputs":2,"x":1060,"y":780,"wires":[["8771abf3.cde338"],["ca60bf95.c9be1"]]},{"id":"251d7643.0d5c4a","type":"switch","z":"d7684e86.ed817","name":"testDir","property":"ikeaDimDir","propertyType":"flow","rules":[{"t":"eq","v":"1","vt":"num"},{"t":"neq","v":"1","vt":"num"}],"checkall":"true","repair":false,"outputs":2,"x":1050,"y":880,"wires":[["79af0635.e08e68"],["461f5f5b.8737f"]]},{"id":"461f5f5b.8737f","type":"change","z":"d7684e86.ed817","name":"Dir Up","rules":[{"t":"set","p":"ikeaDimDir","pt":"flow","to":"1","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":1210,"y":900,"wires":[[]]},{"id":"79af0635.e08e68","type":"change","z":"d7684e86.ed817","name":"Dir Down","rules":[{"t":"set","p":"ikeaDimDir","pt":"flow","to":"0","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":1220,"y":860,"wires":[[]]},{"id":"b8b6790e.53c8c8","type":"change","z":"d7684e86.ed817","name":"Readable","rules":[{"t":"change","p":"payload","pt":"msg","from":"0","fromt":"num","to":"Off","tot":"str"},{"t":"change","p":"payload","pt":"msg","from":"1","fromt":"num","to":"On","tot":"str"},{"t":"change","p":"payload","pt":"msg","from":"2","fromt":"num","to":"Toggle","tot":"str"},{"t":"change","p":"payload","pt":"msg","from":"3","fromt":"num","to":"Hold","tot":"str"},{"t":"change","p":"payload","pt":"msg","from":"4","fromt":"num","to":"Ramp","tot":"str"},{"t":"change","p":"payload","pt":"msg","from":"5","fromt":"num","to":"Direction","tot":"str"},{"t":"change","p":"payload","pt":"msg","from":"6","fromt":"num","to":"Release","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":560,"y":740,"wires":[["c2850c8c.6ab43"]]},{"id":"2eff57e0.9d9128","type":"mqtt in","z":"d7684e86.ed817","name":"Kitchen Bottom","topic":"cmnd/kitchen_button/bot","qos":"2","datatype":"auto","broker":"4802971f.02c578","inputs":0,"x":160,"y":740,"wires":[["3d9e4758.70a708"]]},{"id":"3d9e4758.70a708","type":"change","z":"d7684e86.ed817","name":"To Number","rules":[{"t":"set","p":"payload","pt":"msg","to":"$number(payload)","tot":"jsonata"}],"action":"","property":"","from":"","to":"","reg":false,"x":370,"y":740,"wires":[["b8b6790e.53c8c8"]]},{"id":"7b4976d0.d59648","type":"api-current-state","z":"d7684e86.ed817","name":"Light Status","server":"3503d7cc.127568","version":3,"outputs":1,"halt_if":"","halt_if_type":"str","halt_if_compare":"is","entity_id":"light.ikea_right","state_type":"str","blockInputOverrides":false,"outputProperties":[{"property":"payload","propertyType":"msg","value":"","valueType":"entityState"},{"property":"data","propertyType":"msg","value":"","valueType":"entity"},{"property":"topic","propertyType":"msg","value":"","valueType":"triggerId"}],"for":0,"forType":"num","forUnits":"minutes","x":1390,"y":760,"wires":[["a3c626f4.271b78"]]},{"id":"a16a4dcf.d9542","type":"api-current-state","z":"d7684e86.ed817","name":"Light Status","server":"3503d7cc.127568","version":3,"outputs":1,"halt_if":"","halt_if_type":"str","halt_if_compare":"is","entity_id":"light.ikea_right","state_type":"str","blockInputOverrides":false,"outputProperties":[{"property":"payload","propertyType":"msg","value":"","valueType":"entityState"},{"property":"data","propertyType":"msg","value":"","valueType":"entity"}],"for":0,"forType":"num","forUnits":"minutes","x":1390,"y":800,"wires":[["f65351ea.0824"]]},{"id":"a3c626f4.271b78","type":"function","z":"d7684e86.ed817","name":"Ramp Up","func":"if (msg.data.state == \"off\") {\n    brightness = 0;\n    msg.data.state = \"on\";\n}\nelse brightness = msg.data.attributes.brightness;\n\nif (brightness == 255) brightness = 240;\nelse if (brightness > 220) brightness = 255;\nelse brightness += 25;\nmsg.payload = brightness;\nreturn msg;","outputs":1,"noerr":0,"x":1580, [... cut ...] ome Assistant","version":2,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":false,"cacheJson":true,"heartbeat":false,"heartbeatInterval":30},{"id":"4802971f.02c578","type":"mqtt-broker","name":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]

Mopfus, 2022-03-27:
Hi @neilvangeffen 
Thank you for your approach! I will test it.

One problem I have so far. How can I reach the Button that is connected on my ESP on that I am running WLED? I can't manage this.

Yes, it is a bit of a pity that this function cannot be programmed directly locally on the ESP. The indirect route via HomeAssistant is a bit overkill. But okay. If there is no other way.

If you want it locally, you can try it with ESPhome. If you programm it there, it should work without connection to HomeAssistant.

Subdriven79, 2023-06-03:
Curious if this ever got implemented?   Dim up and down with button press.  I have 2 button 1 rotary knob control box setup and was wanting the rotary to do the color pallets since that wasn't an option with button press either, or is it?

So..  is there brightness adjustment or Color pallet options for button presses?