wled/WLED #201: Long data wire results in random color flickers

gilshahar7, 2019-08-14:
Hey Aircoookie, first of all thanks for the software!

I know this is more hardware based but i figured someone might help.

I am doing an ~8 meter 246 led long strip and i am trying to place the controller at around 2 meters away from the start of the strip. 

I am using a 5v 15a power supply for the strips and a small phone charger (5v 2a) for a NodeMCU.

I connected the data wire of the first pixel along with the ground of the strip (both 22awg  & 2 meter long) to the NodeMCU but colors appear to be random flickers.

When i eliminate the 2 meter wire and connect them at close distance its all normal.

I am using a level shifter to get 5v logic from 3.3v which works normally when im close to the first pixel.

Any idea on how to fix this issue? 

Thanks for any replies.

brentbrooks70, 2019-08-14:
I have a similar setup with about 2.5M from MCU to first LED of 680 over 11.4M. I had some flicker issue not brought on by grounding issues (which you should double check), I was able to solve the issue by placing 1 LED connected almost directly to the MCU and then connecting the cables that had been connected to the MCU from the strip of LEDs to that single LED’s out. After this you you should check the box on the bottom of the LED Setup page “Skip first LED”, you have now used a single LED as a signal booster basically. That resolved the issue for me, hope it helps you as well.

gilshahar7, 2019-08-14:
That's very interesting, ill try that!

Just making sure, you connect mcu data, ground and 5v with the single pixel and then from the pixel you just connect data and ground to the strip?

brentbrooks70, 2019-08-14:
I use a 5v power supply so I connect:
5v to MCU VIN and 5v in on 1st LED
Gnd to MCU ground and gnd on 1st LED
Data Pin on MCU (in my case D4 on NodeMCU without DMA, you have few enough LEDs to use DMA if you like) to DI (data in) of 1st LED, then I run 3 conductor (red, black, green) 22AWG directly to the strip of LEDs. I connect all 3 wires between first LED and the strip. My “strip” is actually composed of 4 strips connected in each corner of the room where I have 12AWG 2 conductor solid wire to re-inject power at the beginning and end of each strip.  I hope that helps make it a little clearer.

gilshahar7, 2019-08-15:
Okay, i just finished wiring it up. the only difference is i didnt connect the first led 5v to the strip 5v since i run it on a different power supply.

Anyways, it works better than it was with the level shifter, colors show normaly but every second there is 1 to 2 flickers. especially at high brightness and blue colors (i think).

I filmed a video to demonstrate what i did. the two long wires that are connected from the first led to the strip are data and ground.

Also i noticed that when i touch the strip with my palm, the flickering goes away immediately (also shown in the video). i wonder if i am missing something? i have seen people use a resistor at the end of the data wire but i have no idea how to find which resistors work here.

Video link: https://youtu.be/gY9cGdvwSEI
Sorry for the dirty soldering, i suck at it :P

Thanks again for the help :)

gilshahar7, 2019-08-15:
Update: i added a 100ohm (i think) resistor to the beggining of the strip data and it got better but it was still flickering at high brightness. 

It seems like an issue with the 2 meter ground wire from the strip to the first led, when i eliminate it and connect it at close range but i keep the 2 meter data wire it works fine without any flickering.

What can i do to fix this?

codamo35, 2019-08-15:
If your power supply has an attenuation dial (assuming so since it's 5v 15a), make sure you are getting the correct voltage at your first LED by using a multimeter at that point. I had this exact issue, where it would occasionally flicker at lower brightness and would go crazy at higher levels. My power supply was reading 12v exactly directly at the power supply, but by the time it got to my first LED (only a feet) the voltage was off a bit (I can't remember if I ended up raising the voltage or lowering it at the power supply. that will probably change for you anyways).

Out of curiosity, why did you end up powering your NodeMCU from a separate charger?

gilshahar7, 2019-08-15:
Well, i power the strip from the middle of it. it reads 5v right at the power supply, but i have a 2 meter wire going up to the middle of the strip, the voltage there gets lower when the strip brightness is high. so i get around 4.2v or even less.

I was planning to connect the nodeMCU to one of the ends but at high brightness, the voltage is just too low for it and it powers off. so i want to power it from a separate power supply (a phone charger) and have 2 meters of data wire and ground wire to the start of the strip.

Not sure why i get voltage drops at high brightness, i use a 1.5mm cable (around 16 awg i think) from the power supply and i connected it every ~2.5 meters or so to the strip for power injection, also at the ends of the strips.

codamo35, 2019-08-15:
Sounds like you might be underpowered with current. What's the calculated power draw for your entire strip (you mentioned 8 meters / 246 LEDs)? With 5v 15a at 80% you should be trying to stay under 60 watts or so.

I also looked back at my notes, and it turns out I had to **reduce** my voltage to fix my LEDs flickering at higher brightness levels. It was set to 12vdc exact, lowered it (-2% to 11.76vdc) and performance drastically increased. Not sure if this helps you but might help someone else searching for this down the line. Flickers are so multifaceted!

gilshahar7, 2019-08-15:
The recomended power supply for 240 leds is 15a, although the flickering happens even when i am only using 5 amps.

brentbrooks70, 2019-08-15:
Do you happen to have another strip you can try? I had to replace 1 section of my 4 section strip as it was causing lots of flickering but only on that segment. Ended up one of the drivers was bad and causing that entire section, the last in series, to flicker. Only reason I suggest this is if you touch it and it changes perhaps one of the drivers is incorrectly mounted on the PCB or there is a crack in the PCB on one of the three lines (5v, gnd, data) that is corrected when you touch or move it.

gilshahar7, 2019-08-15:
I have another 60 led per meter 4 meter stip that i can try but it will take me some time to set up the wiring.

Also, touching the strip is not moving anything physically, i found out that even if i touch my finger on the connections of the start of the strip (or anywhere else) it also fixes the flickering. like my body is acting as a capacitor for the strip or something.

gilshahar7, 2019-08-15:
Just found out another thing. i swapped my 5v 15a psu with a 5v 8a psu that i had lying around.
It seems like there are no issues with it at all..
Maybe a crappy 15a psu? Is it possible?? it works fine when the controller is in close range but not when its 2 meters away?

Anyways, can anyone recommend a power supply that is suitable for this that looks like a laptop ones and not like those industrial ones made from metal?

brentbrooks70, 2019-08-15:
I have several Altove 15A (laptop style) 30A & 60A (industrial metal ones) and have never had any problems. They are available on Amazon and I would imagine other sites. The industrial metal ones have the advantage of being adjustable, I run mine about 5.2 to help with voltage drop

gilshahar7, 2019-08-15:
Actually i just checked, and the 15a power supply that i have is from Altove.. 

This is so weird.. im not sure if the power supply is actually the issue here.. maybe things get crapped out when the current gets high?

gilshahar7, 2019-08-17:
> I have several Altove 15A (laptop style) 30A & 60A (industrial metal ones) and have never had any problems. They are available on Amazon and I would imagine other sites. The industrial metal ones have the advantage of being adjustable, I run mine about 5.2 to help with voltage drop

I found these boost/buck converters: https://www.ebay.com/itm/283573836495 which can accept an input of 1.8v-5v and output 3.3v, will this work for my issue with powering the nodeMCU at the end of the led strip with the voltage drops? At the end of the strip I measured 5v when the leds are off and around 3.5v when they are almost at full brightness.

Sorry if its a dumb qustion, im a noob with electronics.

timothybrown, 2019-08-17:
Yes, I’d pop a small buck/boost converter on to power the ESP. You don’t need a big one, 500mA would be more than enough! (An ESP8266 shouldn’t pull more than 250mA full tilt, running less than 100mA on average.)

The problem is that the NodeMCU board’s use what’s called a “Linear Regulator” to convert the 5V to 3.3V. These require at least a diode drop (about 0.6V) of overhead voltage. I.e., if you want 3.3V out you need at least 3.3+0.6=3.9V in.

Make sure you hook the output of the buck/boost to the 3.3V pin on the NodeMCU and *not* the VIN pin. Also, make sure you *test the output* of the converted with a multimeter first, before hooking it to your board! The 3.3V rail should be 3.3V +/-0.1V.

Alternatively, you could ditch the NodeMCU board and go with something like a D1-Mini, which uses a buck converter instead of a linear regulator. Most buck converters will take an input voltage right down to the output voltage (essentially they run in pass through mode).

Source: I’m a Hardware Engineer. :)

gilshahar7, 2019-08-17:
I ended up getting the 5v boost buck converter. I'll try it out when it arrives.

But in the meantime i do have a d1 mini, you say i can hook up the 5v pin of the d1 mini and it should work even when the voltage drops to 3.3v?

richyxi, 2019-08-17:
https://oshpark.com/shared_projects/kM1DRLS1

This Is a signal Booster, which they used for a similar proyect, u may
check this

timothybrown, 2019-08-17:
Ah, I see, it was a 5V buck/boost, sorry, was having trouble getting eBay to load and assumed it was a 3.3V (or adjustable) converter. A 5V one will work just fine as well!

While I haven’t tested it, the D1-Mini *should* work down to around 3.3V on the VIN. A buck converter works by rapidly switching the the input voltage on and off, think of it as a form of PWM. Since the duty cycle controls the output voltage, turning the switch on 100% of the time will *ideally* pass the input through to the output (minus a tiny bit of voltage drop from the switch).

This is a very simplistic explanation, but it gets the idea across. Also this can vary depending on what switching controller is used in a particular circuit. That said, most of them tend to be able to get within 0.1V-0.3V of the input, which is a lot better than the 0.6V-1.2V of a linear regulator, so it’s well worth a try!

An ESP8266MOD has an absolute minimum input of 3V according to the datasheet, so even a 0.3V drop should be acceptable.

I happen to have a set of D1-Minis coming from Amazon tomorrow, so I think I’ll test it out and see, because now I’m curious!

gilshahar7, 2019-08-17:
Awesome thanks a lot, I'll so try the d1 mini tomorrow and update here.

gilshahar7, 2019-08-18:
Okay, the D1-mini is more resiliant to voltage drops. it does not turn off when the strip is at full brightness white which is great for me :)

Thanks again for the help, now i can power everything from the main psu and no long wires except the button that ill add.

closed 2019-08-18 by gilshahar7

Mariu86, 2019-08-18:
Hello , #gilshahar7 can you please share some pictures with your setup? Thanks

gilshahar7, 2019-08-18:
Definatly, first here is a schematic: https://i.imgur.com/NWNJIth.jpg (the nodeMCU is now replaced with a D1-mini).

And here are some actual photos: https://imgur.com/a/iZMX4QS

Let me know if there is anything i can improve, currently i used hot glue and electric tape to insulate the power injection cuts. the wires are twisted and soldered too.

timothybrown, 2019-08-18:
The only thing I’d change is maybe using crimps to join individual wires together, instead of twisting and soldering them. Solder tends to break very easily, whereas a crimp physically fuses the wire to the metal barrel inside the crimp. You can buy a box of tubular wire to wire crimps and the pliers at any hardware store and the connection will be much more robust.

That said, if this is a permanent indoor installation, twist+solder should be OK. (I still do this from time to time, I just add heat shrink tubing with a bit of hot glue injected in, to seal it and give it extra mechanical support.)

For an outdoor installation I would use crimps with heat shrink tubing over them, and I’d squirt in a little dielectric grease before shrinking the tubing. (This will keep water out.)

gilshahar7, 2019-08-18:
Thanks for the input, this will be an indoor setup but i think i will follow your advice and crimp the wires.

Rajeev-Chandrasekharan, 2021-07-09:
Hello,

I am facing a different kind of issue. 

Currently using a 60 pixel 1mtr strip with WeMos D1 mini flashed with latest WLED firmware.
Both Strip and WeMos is powered through same power supply. Between strip and controller there is 10cm wires connected. Output of the SMPS is connected to 2200MF capacitor.Not using any Level shifter in the output of WeMos 

This setup works perfect and stable when powered through power bank(used for mobile charging) where the transition of effects is smooth and normal.
But all becomes flickering when connected through SMPS power supply(5V 10A). We can observe the difference when power is switched off and back again. Instead of solid color, LEDs start flickering and cycles through different colors. But when the Brightness is set to maximum through WLED interface, it becomes stable. 
It also becomes unstable when switched between various effects.

For example effects like fireworks,Android,Breath,bpm are very unstable on my Pixels.

Do you have some recommendations to fix the issue?

Do you have some suggestions to solve the issue?

Rajeev-Chandrasekharan, 2021-07-10:
Hello,

Just thinks this has been figured out.Changed SMPS to meanwell 5v 15A,all effects and transitions are working great.