wled/WLED #203: WS2811 working fine except incorrect colors

stagea09, 2019-08-18:
uploaded, running perfect with 12v WS2811 strip. but when selecting blue - its actually green, and red is actually blue, and green is actually red.
i see the RGBW setting for adjusting the config but these are like the WS2812b just 12v, and obviously in a different configuration.
is there a way to change the way the controller sends the colors on my strip?

Aircoookie, 2019-08-18:
Hi and welcome to WLED!
Below the RGBW setting there is a "Color order" dropdown field. Try selecting "RBG" and you should be getting the correct colors!

stagea09, 2019-08-18:
Thanks, i was under the impression that the setting was for when you're using RGBW strips. i'll give it a go when i get home and see how i go

stagea09, 2019-08-19:
Yep.. I feel Abit retarded right now.... Thanks ;)

closed 2019-08-19 by stagea09