wled/WLED #205: Bug? cannot find: sdkconfig.h

Mattallmighty, 2019-08-21:
Included NeoPixelBus library by Makuna (2.5.0)
Latest master WLED

Error: No such file or directory: sdkconfig.h 

```
 Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (1M SPIFFS), v2 Higher Bandwidth, Disabled, None, Only Sketch, 921600"

In file included from D:\Google Drive\Projects\Programming\13-08-2019\WLED-master\wled00\wled00.ino:1:0:

C:\Users\Matt\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.h:26:23: fatal error: sdkconfig.h: No such file or directory

 #include "sdkconfig.h"

                       ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

then tried: NeoPixelBus library by Makuna (2.4.3)
also same issue, anyone else experiencing this issue? 
How did you overcome this?

Mattallmighty, 2019-08-22:
Tried on a new PC that had not had Arduino previously installed, followed wiki and same issue.

brentbrooks70, 2019-08-22:
I may be an idiot to ask but why is your sketch not located in your arduino folder? Could you move it there and try again? Also, I have always been told (right or wrong) that library folders should not contain the **-master** on the end, could you try renaming that folder? 

Something else that is very odd, when looking at my **AsyncTCP.h** (or **AsyncTCP.cpp** for that matter) it does not reference **sdkconfig.h** anywhere. and the only references I find when Google-ing this is when trying to load an ESP32 board with ESP8266 flash settings (or the other way around). Are you using the right library and config for your board?
You may also want to look at the last entry for [this](https://github.com/nkolban/esp32-snippets/issues/253) issue.

FYI - I am somewhat a newb to all this and I could be way off base in which case maybe one of the heavy hitters for this project will weigh in with some good info.....

timothybrown, 2019-08-22:
Yes, my research indicates @brentbrooks70 may be on the right track here. Specifically the issue he linked, which I’ll quote here:

> In case anyone else is having this problem, there is apparently a conflict between the expressif ESP32 board libraries and the ESP8266 community ones. Removing the latter fixed the problem.

I’m unable to reproduce your issue in the Arduino IDE. Can you share *exactly* what libraries you have installed?

timothybrown, 2019-08-22:
Actually, after thinking about this for a minute and it occurred to me something wasn’t quite right, then I realized! AsyncTCP is a library for the ESP32! *ESPAsyncTCP* is the version for ESP8266 Arduino.

ESP32: https://github.com/me-no-dev/AsyncTCP
ESP8266: https://github.com/me-no-dev/ESPAsyncTCP

So, yeah, you’re running the wrong library.

Mattallmighty, 2019-08-27:
Hey team,
Legends, realised my chipset error. @timothybrown you are right, I cleared all my existing libraries, installed into my library the Esp8266 and this has fixed the issue.
Thanks to both of you for your input. :)
Have a great day

closed 2019-08-27 by Mattallmighty