wled/WLED #208: Expose the COOLING option via API and possibly GUI

brentbrooks70, 2019-08-24:
Anyone else interested in this or is it just me?

Aircoookie, 2019-08-24:
I agree that the COOLING options should be accessable both via the APIs and GUI.
Since it just affects a single effect though (Fire 2012), I don't believe it is justified to add an extra FX parameter, because it might confuse users more if there are more settings. Rather my proposal would be to have the speed value change the cooling. Right now, speed does work, but only changes the framerate of the animation. This is something I am transitioning away from either way (making all effects run at a constant 50 or so fps) because only a fixed framerate allows me to perfect the segment feature and allow new features such as smooth transitions from one effect to another.
What do you think?

sansillusion, 2019-08-24:
That sounds great !

brentbrooks70, 2019-08-24:
That's awesome!!! Thanks

Aircoookie, 2019-08-24:
Great! No, don't get me wrong :) I'm not getting rid of speed at all, constant FPS just means that speed is implemented in a different way (a counter that keeps track of FX progress)

I'll push a small update with the cooling changable tomorrow!

eoncire, 2019-09-18:
> I agree that the COOLING options should be accessable both via the APIs and GUI.
> Since it just affects a single effect though (Fire 2012), I don't believe it is justified to add an extra FX [...]

I've used another sketch which you pulled the Twinklefox effect from.  The way it handled the Fire2012 effect was brilliant and very easy to tune.  If i recall correctly, the speed slider basically was the amount of "fire" that was calculated, and the intensity slider controlled the cooling.  It was easy to tune the effect to the length of the LED strand to make a very nice looking fire effect.

closed 2019-10-02 by Aircoookie