wled/WLED #209: Load specifc color or effect when switch on

Dgemily, 2019-08-24:
thank you for this software. I definitly chosse it for my lamp project !
Is there a way to always load the same color/effect when we turn on ? ( I don't wpeak about the boot, but when we turn off/on from interfacte or button.)

Aircoookie, 2019-08-24:
Hi,welcome to WLED!
Currently that is not possible when turning on via the web interface.
(but I could add an option to always switch to preset X when turning on!)

For the button however, you could setup an [API macro](https://github.com/Aircoookie/WLED/wiki/Macros) (like `PL=1`) which loads a preset when turned on and turns off when long or double presses (`T=0`).

Dgemily, 2019-08-24:
Hi, thank you so much.

it can be a really nice option to use preset !
and yes about button, it so nice to be able to parameter it with macro. but I already use long press to load a presets cycle and double clic to swicht between presets or effects (not sure yet what I will do).
I use :
short press: `T=2`
long press: `CY=1&PX=1&PT=15000&P1=1&P2=25 `
and double clic: `FX=~`

so just need to load a specific preset when sticth on and it will perfect!

thank you again for your great software!

closed 2019-10-25 by Aircoookie