wled/WLED #210: New binaries with latest updstes

chitrangad, 2019-08-24:
Hi would it be possible to release new binaries with all the latest updates (mqtt, cooling etc)? 
Thanks for the great work and awesome software!

Aircoookie, 2019-08-24:
You are correct, this is long overdue. Releasing 0.8.5 tomorrow!
You're welcome! :)

Aircoookie, 2019-08-27:
Takes a bit longer due to a short-term issue! Sorry :)

alanlawton, 2019-09-09:
Has this been released somewhere?

Aircoookie, 2019-09-09:
Here is a binary of current master!
Still need to do some bug testing before releasing 0.8.5, coming this week!
https://cdn.discordapp.com/attachments/473448917543944193/620321838169325598/wled_1908303_master_ESP8266.bin
https://cdn.discordapp.com/attachments/473448917543944193/620329347601006632/wled_1908303_master_ESP32.bin

closed 2019-10-02 by Aircoookie