wled/WLED #211: SPIFFS configuration file

Aircoookie, 2019-08-27:
WLED has SPIFFS support, it can be enabled via headers in wled00.ino. I considered switching to an SPIFFS configuration file and will likely do soon-ish. The main problem would the high memory usage of using descriptive JSON keys. However I believe it would drastically increase transparency and code legibility. Do you know whether it's possible to have F() (or similar, store in flash) strings as JSON keys with ArduinoJSON 6? In that case, i'd highly prefer to switch to a config file in SPIFFS. Another issue is cleartext passwords, I'd rather not store them in an accessable file (at least not the home WiFi password). Maybe not storing them at all explicitely and making use of WiFi.persistant() would work, so we could get rid of the current EEPROM implementation entirely.

theshadowsvk, 2019-08-27:
Hello Aircoookie :)
If I understand correctly what you mean, try to take a look on https://github.com/toblum/McLighting.
You probably already know about this project, and I studied this project a while ago, and he is using SPIFFS to store json config file right there. He also using JSON to handle everything, including the webUI. Maybe you can switch from xml to json this way in future also ;)

Aircoookie, 2019-08-30:
Thanks for all the detailed insights! No, I don't think that we have to worry about a flash dump too much - the risk for that is even lower with SPIFFS then with EEPROM as the exact memory location in SPIFFS is potentially harder to figure out. I should have been clearer about what I mean, which is not SPIFFS per se but rather a built-in "file browser" function (which can be enabled in wled00.ino with `#define WLED_ENABLE_FS_EDITOR`), which would allow the user direct access to the contents of the filesystem (which I believe is useful not only for debugging, but also for manually making changes to presets, playlists and palettes that will be saved there in the future) which will also contain the `config.json` with all the settings.

That said, access to this editor is disabled if the OTA lock feature is set. I believe we could improve security by activating the OTA lock by default again - just need to figure out how to set an unique password by default that is not based on the MAC... will look into that in due time!

About the PSK, that would certainly be an option (although I couldn't find a function to connect to a network using the PSK, `WiFi.psk()` is just for obtaining it). We'll see, but I believe the OTA lock is secure enough, given that there are other potential attack vectors regardless, like an attacker connecting to WLED-AP and obtaining the password from the unencrypted POST request that is sent after the user saves the WiFi settings.

timothybrown, 2019-08-31:
@Aircoookie Ah, I understand, sorry about that, the file browser didn’t even occur to me. (Like I said, I spent a few years in network security so my mind immediately starts processing all the scenarios and I can go off on a bit of a tangent...) Anyway, the file browser would still require someone to already have direct access to either the device or network, so I agree with your thoughts, especially in light of the OTA lock function.

As for the PSK, I believe you can simply use it in place of the passphrase with the standard WiFi connect function. I’ll double check that tonight as it applies to the ESP8266 WiFi library, but I know this is the case under other platforms (TI’s CC3200, Linux, macOS/iOS) as it’s part of the WPA specification. Going by the standard, it is the *technically* correct method to use (calculate the PSK once from the passphrase, then only store and use the PSK). In practice I’m not sure it really matters that much either way. It might speed up the connection process by a small amount (maybe *a* second) as the device doesn’t have to generate the key each time.

blazoncek, 2022-02-21:
Closing due to LittleFS being implemented since 0.11 and replacing EEPROM use.

closed 2022-02-21 by blazoncek