wled/WLED #212: Hyperion Config?

OliPassey, 2019-08-28:
Hey - i have WLED working fully, and attached to the back of my tv - but i can't get hyperion to connect to it. Would anyone who has got this going mind taking a look at my config?

hyperion.config.json:
https://gist.github.com/OliPassey/691fc97d30e759ba3b4330071994f4e3

My WLED is running on 10.0.0.97 - and is reachable by the raspberrypi running kodi / hyperion.
(did a ping test)

In the hyperion logs i see:
```
ssh out: 	/storage/hyperion/bin/hyperion-remote.sh  -c 855dff
ssh in: 	hyperion-remote:
ssh in: 		version   : V1.03.5 (GitHub-66bef6b/fb413cd-1566231780
ssh in: 		build time: Aug 25 2019 10:39:08
ssh in: 	Connected to localhost:19444
ssh in: 	Set color to 133 93 255
```

I presume the localhost:19444 is hyperion connecting to something local which then forwards onto the WLED IP as detailed in the top of the config file:

`	"device" :
	{
		"name"       : "MyHyperionConfig",
		"type"       : "udpraw", // also tried regular udp
		"output"     : "10.0.0.97:19446",
		"rate"     : 1000000,
		"maxpacket"     : 1450,
		"protocol"     : 0,
		"colorOrder" : "grb"
	},`

But the LEDs don't light up. WLED is able to control them from the GUI - but not Hyperion.
Am i missing something from the forwarders section?
```

	// JSON SERVER CONFIG
	"jsonServer" : 
	{
		"port" : 19444
	},

	// PROTO SERVER CONFIG
	"protoServer" : 
	{
		"port" : 19447
	},
```

Any help would be greatly appreciated

OliPassey, 2019-08-28:
Always 5 minutes after asking for help....

Removing these lines from the config and now all is well.
	// JSON SERVER CONFIG
	"jsonServer" : 
	{
		"port" : 19444
	},

	// PROTO SERVER CONFIG
	"protoServer" : 
	{
		"port" : 19447
	},

Updated the gist for anyone looking in the future.

sjude68, 2019-09-04:
I am using the Hyperion capture app on my Sony Android TV  and Wled.  What config changes should I do to make Hyperion work.

https://play.google.com/store/apps/details?id=nl.hyperion.hyperionfree

OliPassey, 2019-09-04:
When you say Hyperion Capture App, do you mean the one you've linked to?
That app doesn't do screen capture, and it connects to the Hyperion Daemon that you need to have running in your network. I have mine on a RaspberryPi that also runs kodi, doing the screencapture, it then forwards the lighting commands onto WLED via UDP.

Mariu86, 2019-09-04:
It's a way to screen capture the Netflix?

Aircoookie, 2019-09-05:
It may not be possible to make it work with Netflix due to anti-DRM measures, but using it in a browser on a (Linux) computer might be your best bet. 4K won't work for sure, though.

phamre, 2019-09-05:
> When you say Hyperion Capture App, do you mean the one you've linked to?
> That app doesn't do screen capture, and it connects to the Hyperion Daemon that you need to have running in your network. I [...]

i think me means this app..

https://play.google.com/store/apps/details?id=com.abrenoch.hyperiongrabber

sjude68, 2019-09-05:
Yes this app also does screen capture and forwards to the IP:port specified

sjude68, 2019-09-21:
I removed the lines suggested by https://github.com/OliPassey but still the WLED does not play according to the Hyperion on Raspberry Pi.  I am  using the Grabber software on my Android TV which is configured to use Hyperion on Raspberry Pi.  Raspberry Pi is connected by USB to Arduino Nano to the WS2812 LED strip.  I have another LED strip connected to WLED on Nodemcu.

How do I configure WLED to play Hyperion instead of using the LED strip on Raspberry Pi.

[hyperion.config.zip](https://github.com/Aircoookie/WLED/files/3638493/hyperion.config.zip)
My hyperion.config.json file on the RaspberryPi is attached.

tpmodding, 2020-01-06:
in hyperion.ng you can choose the e1.31 device, and at hyperion you can choose the udp device

as above mentioned you have to put in the WLED IP at the Output part

"device" : {
"name" : "MyHyperionConfig",
"type" : "udpraw", // also tried regular udp
"output" : "192.168.1.97:21324", <-----ip and port from your wled device
"rate" : 1000000,
"maxpacket" : 1450,
"protocol" : 0,
"colorOrder" : "grb" },

piowys, 2020-01-12:
hyperion.ng controller type should be **udpraw** and you should use your WLED IP as Target IP and port **19446** as stated in documentation: https://github.com/Aircoookie/WLED/wiki/E1.31-&-UDP-Realtime-Control#hyperion (I've UDP realtime enabled and E1.31 multicast disabled in WLED sync settings).

Nevertheless, I can't find this 19446 port in WLED settings anywhere, @Aircoookie IMO it should be at least pointed/described somehow in the sync settings page.

And thanks for WLED! It's great! ;-)

daekka, 2020-01-18:
I'm not able to connect from android hyperion apps. 
I use WLED version 0.9.0-b2

preslav87, 2020-01-19:
I have also the same issue. All the hyperion apps from the market can't connect to the wled. I had tried the 19446 and no effect..
I am using the wled version 0.9.0-b1

sjude68, 2020-01-19:
Hyperion app doesn't connect to wled directly. The app captures the screen and sends to hyperion server. Hyperion should be running on Raspberry Pi. The server forwards to wled.

recliq, 2020-01-19:
Just to expand on @sjude68 comment:

Hyperiond (the server component) accepts two types of input on two different ports, by default these are 19444 and 19446.
Port 19444 accepts JSON messages and is used by the eg. the adroid hyperion remote app.
Port 19446 accepts binary data and is used to receive the captured image data.

WLED understands neither of these, instead it can be used by hyperiond as an output device (udpraw protocol on port 19446).

The use of port 19446 for updraw protocol on the WLED side might be a bit misleading...

sjude68, 2020-01-19:
So 1st setup Hyperion server on Raspberry Pi or any other device. Then instead of using the led strip directly connected to the server, use wled connected led strip.

When hyperion server is off the strip works as configured on wled.

shafr, 2020-04-15:
My config is: 
* Android TV with `Hyperion Android Grabber` to capture screen and send it to Hyperion
* RPI-compatible device with `Hyperion NG` that can act as server 
* WLED with ESP8266 that acts as output.

Was anyone able to make it work and can provide reasonable manual on that ?

austwhite, 2020-04-15:
DrZzs did a video of hyperion a little while ago and he used a WLED module.
It works well.
I have done it as well.  
Hyperion can use E1.31 but it can also send out raw data.
WLED can accept both of these

austwhite, 2020-04-15:
In Hyperion under Configuration and LED hardware, setup your Controller Type as upde131 or updraw depending on your preferred method.  Default ports should work.   Target IP is the IP address of your WLED board.
E1.31 uses port 5568.  Can't remember off hand the raw port.
From there it's pretty well the same as any other Hyperion setup 🙂