wled/WLED #213: On selection of red color Wemos Hangs . Requires Hard reset to start back

sujitrp, 2019-09-03:
Version WLED @version 0.8.5-dev Date  30 ‎August ‎2019, ‏‎01:38:07 IST 8:00 pm GMT
arduino IDE : arduino-1.8.7
All Library are up to date . 
Hardware : Wemos Mini 
                  LED strip : 5M WS2812B 5 Pins RGBWW 4 IN 1 LED Strip Light Non-Waterproof DC5V - RGB+White 1M

https://www.banggood.in/1M-5M-WS2812B-5-Pins-RGBW-RGBWW-4-IN-1-LED-Strip-Light-Non-Waterproof-DC5V-p-1211152.html?rmmds=myorder&ID=52881245763&cur_warehouse=CN

[image]

Tried Debug Level Core & All option but no entry in log

Aircoookie, 2019-09-03:
Hi! Thank you for the bug report.
Does this only happen when selecting a red color or also sometimes when using other UI functions?
Do you use mobile or classic UI? If mobile, does it happen if using the color wheel or the red button?
You can try setting LwIP variant to "1.4 High Bandwidth", that may resolve some issues.

sujitrp, 2019-09-03:
I tried this using web ui , manually changing the red color slider & also during random color / rainbow automation when it comes to red color ,strip gets red color next second it gets dim and hangs .then post hard restart it starts working .  I shall try using mobile ui and  1.4 high bandwidth and update

sujitrp, 2019-09-04:
Hi 
I tried both option 1.4 and 2,0  but same issue . LED lights turns red then loses Brightness with ESP stops working . either select the Red color using UI classic or Mobile or during automation  sudden it changes to Bright red the loses Brightness 

With Upload Speed 115200
With CPU Frq : 80MHZ
Debug level core
V1.4 Higher Badwidth 
Result : crashes on start  

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:1.4.0rc2/BearSSL:6778687
mode : sta(84:f3:eb:b2:2b:3a) + softAP(86:f3:eb:b2:2b:3a)
                                                         add if0
                                                                bcn 0
                                                                     del if1
                                                                            add if1
                                                                                   dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
                  bcn 100
                         bcn 0
                              del if1
                                     mode : sta(84:f3:eb:b2:2b:3a)
                                                                  scandone
                                                                          state: 0 -> 2 (b0)
                                                                                            state: 2 -> 3 (0)
                                                                                                             state: 3 -> 5 (10)
                                                                                                                               add 0
                                                                                                                                    aid 2
    cnt

        connected with SUJIT24, channel 1
                                         ip:192.168.1.25,mask:255.255.255.0,gw:192.168.1.1
                                                                                          Ada

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:1.4.0rc2/BearSSL:6778687
mode : sta(84:f3:eb:b2:2b:3a) + softAP(86:f3:eb:b2:2b:3a)
                                                         add if0
                                                                bcn 0
                                                                     del if1
                                                                            add if1
                                                                                   dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
                  bcn 100
                         bcn 0
                              del if1
                                     mode : sta(84:f3:eb:b2:2b:3a)
                                                                  scandone
                                                                          state: 0 -> 2 (b0)
                                                                                            state: 2 -> 3 (0)
                                                                                                             state: 3 -> 5 (10)
                                                                                                                               add 0
                                                                                                                                    aid 2
    cnt

        connected with SUJIT24, channel 1
[... 243 lines cut ...]
:urd 1, 94, 77
:urd 5, 94, 78
:urd 1, 94, 83
:urd 2, 94, 84
:urd 2, 94, 86
:urd 1, 94, 88
:urd 1, 94, 89
:urd 1, 94, 60
:urd 11, 94, 61
:urd 1, 94, 72
:urd 4, 94, 73
:urd 1, 94, 77
:urd 5, 94, 78
:urd 1, 94, 83
:urd 2, 94, 90
:urd 2, 94, 92

─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

Session stopped
    - Press <return> to exit tab
    - Press R to restart session
    - Press S to save terminal output to file

Error reading from serial device

Aircoookie, 2019-09-05:
Thank you for the debug logs. I have a suspicion that there is a hardware problem. Can you try disconnecting the LED strip and see whether the software still crashes. Alternatively, you could try a spare ESP if you have one handy. By the way, what power supply do you use? It may be an under-power problem, although that would probably happen with white rather than red color...

sujitrp, 2019-09-05:
Thanks for response. Since 0.8.5 is crashing so  uploaded 0.8.3 on two WEmos . Both are working fine with 0.8.3. No crash . Today almost 4th day it is running 24 *4 hrs .  Yesterday to get Log from same set up , Yesterday  I uploaded  0.8.5  using OTA & USB both method with IDE .  Post of which I reverted back to 0.8.3 and last 20 Hrs 0.8.3 is working without crash.
But yes without LED strip connected Software does not crash . HTTP server keeps on running . But as on when WeMos connected with LED strip it crashes . This happens only on 0.8.5 but It works fine 0.8.3.

closed 2020-04-05 by stale[bot]