wled/WLED #214: Can you provide wiring diagram or guide for RGB 5050 12V led strip?

hristo-atanasov, 2019-09-07:
It would be nice if there is a gude or schematics for connecting/wiring/powering RGB strips and description of the components needed, like the mosfets for example. Thank you in advance. :)

raoulteeuwen, 2019-09-07:
Not sure what others think and what you are looking for, but sounds to me this is outside the scope of WLED. There's lots of info on the web. I just started with LEDs and have blogged the most usefull info i found, maybe that helps you? > https://raoulteeuwen.blogspot.com/2019/08/comparison-of-led-strips.html

hristo-atanasov, 2019-09-07:
@raoulteeuwen I have many projects completed with different types of LED strips. As is in the info of this repository is written it can control non addressable RGB led strips, and I have 20 meters such strip inside my house, I need a help to connect it to ESP8266 and control it with WLED. It is not outside of the scope of WLED.

This is why I opened this "issue". If someone connected 12V RGB strip to ESP8266 and control it with WLED, please give me some detailed info. :)

raoulteeuwen, 2019-09-07:
@hristo-atanasov I did connect a few 12V LED strips to both a NodeMCU and a Wemos D1 mini and put WLED on both ... works fine. But my strips are adressable, so i guess not what you're looking for. If so: sorry bothering you.

theshadowsvk, 2019-09-07:
@hristo-atanasov No, WLED is only about addressable leds. You must misunderstood this project.

Aircoookie, 2019-09-07:
Hi, yes this is correct, this project is made for addressable WS2812 LEDs. 5050 are not officially supported by me. There is an issue thread about this, https://github.com/Aircoookie/WLED/issues/58, it is possible, but I won't provide support on it. I highly recommend investing in an addressable strip, you'll be missing out on a lot otherwise!

hristo-atanasov, 2019-09-07:
@all First, just to clarify 5050 is just a form factor .. WS2812, APA102, and not addressable/analog LED strips are all 5050 - 5.0mm x 5.0mm. :)

RGB strips are actually the same as RGBW, with the small difference that they are missing the fourth separate channel for the additional White diode on the strip. They work on the same principle, just without connecting the W channel.
@theshadowsvk ( without any bad feelings ;) )
When someone tell me that I misunderstood something I get a little grumpy, but just a little.
When all we read the "Features" section of "readme.md" and go to point number 6 we will all read "Support for RGBW strips". 
RGBW strips are 4 colours channel, analog non addressable strips, like RGB strips which are 3 channel, analog, non addressable strips. Now when we all know this information, and we are clear that I haven't misunderstood, what the project is, we can continue the conversation.

@Aircoookie 
I had to read all the information about how RGB, RGBW, RGBCW, RGBCCT work internally, how they can be connected and how they can be controlled. Now I feel smarted than yesterday! :D
I can share information about how we can connect RGB and RGBW strips to NodeMCU and use it with WLED. I will first make a tutorial, and then share it here, so if you want you can officially support it later. :) About the investment in addressable LED strip - before 5 year, when I was renovating my flat I invested in, and installed  35 - 40 meters WS2812 and APA102. Before one week I ordered another batch of 4pc. 5m. WS2812. The thing is that I have one room that I made with infinity mirror effects integrated in the walls and the ceiling and I can just change the controller and the power supply. This is why I'm looking for RGBW controller solution based on ESP8266 .. Just like WLED. I haven't used it yet, but I think it is perfect for all my led strips. Some of the sketches I wrote by myself, but from last year I started using opensource "flash-configure-and-use" firmware. As I see, your software is even more powerful than the one that I'm using, I'm definitely moving on to it. I highly appreciate your work! And from what I'm reading other people too. Great job! :)

hristo-atanasov, 2019-09-07:
I just found interesting video that explains the analog strips, if someone is interested. :)
https://www.youtube.com/watch?v=f_B7X9OLouQ

HimbeersaftLP, 2019-09-07:
> RGBW strips are 4 colours channel, analog non addressable strips, like RGB strips which are 3 channel, analog, non addressable strips. Now when we all know this information, and we are clear that I haven't misunderstood, what the project is, we can continue the conversation.

No, RGB and RGBW just stands for the colours they support, not how they are addressed. This project only supports digitally addressable strips that use WS2811/2/b's protocol.

If you're looking for a digitally addressable RGBW strip, I use SK6812 ones without a problem with WLED.

hristo-atanasov, 2019-09-07:
@HimbeersaftLP The purpose of this post is not to offend each other! Yes, from few years there are addressable RGBW strips, which is cool, but .. It doesn't matter .. Leave it .. Already explained why I opened this "issue". Read everything before answer!

So .. Ok .. This project is not for not addressable/analog strips. Ok! I understand! Is it enough. No we can all come down. Stop acting like I got all of you for hostages. ;) Piece brothers. We are all here to help each others and improve what we can improve. Or it is just me, who is thinking this way. Anyway .. When I first found this project I went trough all Issues, and I saw that someone already have used analog string and this is why I asked. I am not saying there is a problem with the code, or there is a bug .. Just opened this "issue" in case that anyone knows how, to share with others. :) As I said in my previous answer I respect the works of the developer and anyone else that is contributing! :) No bad feelings folks! :) Cheers! :)

HimbeersaftLP, 2019-09-07:
@hristo-atanasov
> The purpose of this post is not to offend each other!

Shut the fuck up, I was not trying to offend anyone until now. Stop trying to act like everyone is attacking you.

> Already explained why I opened this "issue". Read everything before answer!

I did, and apparently more thorough than you.

> PS: WS2811/2/b and SK6812 are not protocols

When did I ever imply that? I literally said "WS2811/2/b**'s** protocol"

If you understand so well that this project is not for analog strips then why are you still commenting, assuming people here don't know what an analogly controlled strip is?
Aircookie already linked an issue with a discussion about how to use those.

hristo-atanasov, 2019-09-07:
@HimbeersaftLP Too much talking, too less working .. Ju/a/erk! :D What are you? lawyer? Secretary? Parrot? When I was programming chips, controllers, ROMs, PROMs, EPROMs and EEPROMs, you probably played on the playground ..

hristo-atanasov, 2019-09-07:
@Aircookie you can now close this issue! :)

closed 2019-09-07 by hristo-atanasov

HimbeersaftLP, 2019-09-07:
Thank you for you completely meaningless answer without any content.
I'm an electronics student and you're most likely right with the assumption that I was a child when you were already into electronics, however that doesn't make you any less of an asshole

hristo-atanasov, 2019-09-07:
HimbeersaftLP 

> If you understand so well that this project is not for analog strips then why are you still commenting, 

Why are you commenting in a closed issue? :D

HimbeersaftLP, 2019-09-07:
@hristo-atanasov Because you asked me a question and as a polite person I, of course, answered it.

briankimstudio, 2020-04-15:
> HimbeersaftLP
> 
> > If you understand so well that this project is not for analog strips then why are you still commenting,
> 
> Why are you commenting in a closed issue? :D

Hi,

I had the same question but couldn't find any solution on the internet. So, I made one. Check it out if you are still interested in controlling an unaddressable RGB strip with ESP32.

https://github.com/briankimstudio/aled

Hope you enjoy it.

Thanks.

briankimstudio, 2020-04-15:
@hristo-atanasov  

Hi,

I had the same question but couldn't find any solution on the internet. So, I made one. Check it out if you are still interested in controlling an unaddressable RGB strip with ESP32.

https://github.com/briankimstudio/aled

Hope you enjoy it.

Thanks.