wled/WLED #215: Having disco lights instead of normal lights

miesjel12, 2019-09-08:
Hi there,

First time WLED user here, so sorry if I'm asking something that has allready been a topic, i just could not find the answer anywhere. 

I've flashed version 0.8.4 (because of the HA auto discovery), unedited. After connecting the GPIO pin to the WS2812 LEDs and turning it on, the only thing i see is each LED showing a random color like a disco. The sliders on the page dont respond and turning off is not possible, it just turns af part of the LED's and the rest continues to stay in party mode. 

I will ad a picture later, my phone just crashed and died :)

For testing, i've connected everything like this:
- Wemos D1 mini on USB port of computer. 
- 5v LED's (116 of them), connected on their own power supply with an output of 5v - 8A. 
- The green wire is obviously connected to the Wemos for signal. 

I've set the number of leds in the settings, as well as the maximum current and try to play with the rgb order, but none of the options make any different. 

As said I will add a picture asap if its not clear enough!

Thanks in advance. 

Michel

theshadowsvk, 2019-09-08:
Are you using BitBang method ? The same thing happened to me when testing neopixelbus, just switched to Uart method(different pin) and perfect :-) with your pixel count, you can go better with DMA method which is more cpu friendly and you have enough ram on your esp8266 for 116 pixels ;-)
Btw did you used level shifter ? Cause this cause sometimes troubles.

raoulteeuwen, 2019-09-08:
@miesjel12 you talk about the green data wire, but do you also have the ground in order? I myself have both a data wire as well as ground running between the LED strip and the Wemos...

miesjel12, 2019-09-08:
> Are you using BitBang method ? The same thing happened to me when testing neopixelbus, just switched to Uart method(different pin) and perfect :-) with your pixel count, you can go better with DMA [...]

I'm using the default bin file, looks like that switches automaticy if i've read it correctly, cant find a way to change it in the software so not sure which one it choose :)

> 
> 
> @miesjel12 you talk about the green data wire, but do you also have the ground in order? I myself have both a data wire as well as ground running between the LED strip and the Wemos...

The ground is going to the power supply, thats taken care of since i've had a lot of problems with other software and grounding (learned from my mistakes!). But thanks for suggesting, a lot of problems are indeed caused by not grounding correctly. 

In the few minutes that you guys replied (thanks for the very quick respons!!) I might have found the problem. I rememberd I had another Wemos d1 mini with a power board so it could connect to the same power supply as the LED. After reconnecting everything, it stopped the party mode and started working as expected!

So in short, the problem was in the power connection, probably inb the USB to the wemos. 

Problem solved for now!

brentbrooks70, 2019-09-08:
It was most likely not just the power supply but you need to have a shared ground. Both the data and ground cable need to be attached to the MCU, now that you are using the same power supply you have a shared ground. If you are interested, try the old board again but this time commect a ground wire from the LEDs to MCU as well as data and see if it works. Good luck!

miesjel12, 2019-09-08:
> It was most likely not just the power supply but you need to have a shared ground. Both the data and ground cable need to be attached to the MCU, now that you are using the same power supply you [...]

Ah okay! Thanks for explaining, this is new for me. I will try and see what happens!

closed 2019-09-08 by miesjel12