wled/WLED #216: Try to use HTTP JSON via Python from Raspberry

Physikus74, 2019-09-09:
Hello,
I try to use the JSON API of WLED from a Raspberry pia via Python script. The necessary "requests-lib" is installed. I get "413" back for all attempts. Can anyone help me?
the content of the script looks like this:
------------------------------------------------------------------------------------------------
#!/usr/bin/env python

import requests
import json

payload = {'bri': '128'}
r = requests.post("http://wled-light/json/state", data=payload)
print r
--------------------------------------------------------------------------------------------

Many thanks in advance,
Physikus 74

Physikus74, 2019-09-10:
Addition:
After some searching in the WWW I have solved a part of the problem. In reply I get "200" back, which means that the message has arrived. However, the values are not changed on the device. Can someone tell me what else I'm doing wrong?
The script now looks like this:
`#!/usr/bin/env python

import requests
import json

payload ={"on":"true","bri":"255"}
headers ={'content-type':'application/json'}
r = requests.post("http://wled-light/json/state", data=json.dumps(payload),headers=headers)
print r.status_code
print r.json()`

Aircoookie, 2019-09-18:
Hi! I'm very sorry, there was a bug that caused the JSON API to ignore requests a few commits back. You can try again with the latest release 0.8.5. Code 413 means that the content length was too long somehow. Great you got that sorted out already. Let me know if it still doesn't work after updating :)

Physikus74, 2019-09-25:
Hello,
Today I made the update to version 0.8.5.
All settings were deleted, including the connection to the WLAN. But the WLED-AP was reachable. I tried to configure SSID and password of the WLAN. After the restart no connection to the WLAN is established. Only the WLED-AP answers. Now I am helpless...

raoulteeuwen, 2019-09-25:
> Hello,
> Today I made the update to version 0.8.5.
> All settings were deleted, including the connection to the WLAN. But the WLED-AP was reachable. I tried to configure SSID and password of the WLAN. [...]

- How did you do the update?
- On what IP was the WLED AP reachable after the update?
- What is the situation after the restart? The same as before (WLED AP reachable but WLED AP has no connection to your WLAN(?

Physikus74, 2019-09-26:
The update was done via "manual OTA".
Because I have hardware with ESP32, I chose the file: WLED_0.8.5_ESP32.bin
After the first restart the IP was 4.3.2.1. Via WEB-UI I entered SSID and password of my WLAN.
After a further restart, WLED reports again with its own AP (WLED-AP) under IP 4.3.2.1. There was no login in the WLAN.
It is also odd that even immediately after the update no password query (WLED1234) was made at the first login. 
Then I tried the "Factory-Reset" and after restart again entered SSID and password of my WLAN... ...all without success...

Physikus74, 2019-09-26:
Postscript:
According to the description, the restart should take about 30 seconds when logging in to the WLAN. However, hardly more than 3 seconds elapse when restarting after entering the SSID and password of the WLAN...

Physikus74, 2019-09-30:
I solved the problem with the WLAN myself:
Not only the configuration of the WLAN didn't work, all other settings weren't taken over after restart either.
I guess that the binary file for direct flashing is not compatible to my ESP32 board. So I recompiled the sources. After that everything worked again, also the settings have been preserved.

Physikus74, 2019-10-21:
Hello,
now that I have version 0.8.5 running, I tried again to send JSON commands to the system via HTTP post. (using Python with requests-lib on raspberry pi) The result is the same:
On a "POST" the acknowledgement gets back with "200", no matter what I send within the payload.
With a "GET" I get the correct state of the queried variables.
Can someone tell me where the error is in my script shown earlier?

Thank you and many greetings,
Physikus

Aircoookie, 2019-10-21:
Hi,
getting back 200 after POST request is expected behavior, you can add `{"v":true}` to your payload JSON however for WLED to return the entire state object again.
If you only want to obtain the current state, always use GET.
I'd need someone more proficient in Python to tell you what's wrong with your script, but it might be that it is trying to send the JSON command in an invalid way. The JSON needs to be converted to a string, I don't know if Python does that automatically.

Greetings and I hope you'll get the script working soon!

Physikus74, 2019-10-21:
Hello and thank you for the quick answer!
In Python I am also a beginner and have now tried to solve the problem for a long time. Now I have succeeded. Maybe I can make the start a bit easier for other beginners. ;o)
The answer is, as so often, quite simple:
In the Wiki you can find an example how the data should look like in post-request: ...{"on":true, "bri":255}...
If you try this under Python 2.7 like this, you get an error message from the interpreter. Then I tried it like `...{"on": "true", "bri": "255"}...` The interpreter was satisfied with that, but WLED could not resolve the sent text. Python 2.7 is case sensitive for boolean variables. The data must look like this: `...{"on":True,` "bri":255}...` is the simple answer. I add a script which works. Now I can try the segment support via JSON. 
`#!/usr/bin/env python

import requests
import json
#Achtung: Boolsche Werte sind Case-sensitive und werden gross geschrieben!
json_data ={'on':True, 'bri':255}
r = requests.post("http://wled-light/json/state", json=json_data)
print r.status_code
print r.json()`

closed 2019-11-12 by Physikus74