wled/WLED #220: canvas addressable pixels

trdb33, 2019-09-14:
Howdy
installed Wled yesterday and i'm very satisfied with all the possibilities, thank you for this amazing work !!
What I'm looking for is the possibility to send to each pixel a color value via mqtt, the idea is to avoid flashing the board each time we want to add a new palette

This is an old Arduino tinker created years ago I'm still using
[https://www.dropbox.com/s/zvtc9p5tojipszw/Ps-canvas.jpg?dl=0](url)

JAMESBOWLER, 2019-09-14:
Segments are individual parts of the LED strip. In 0.9.0 this will enable running different effects on different parts of the strip.

This will be in the next release.

https://github.com/Aircoookie/WLED/wiki/JSON-API

Aircoookie, 2019-09-18:
That's correct! However the Segments won't be ideal for your usecase as there will be a maximum of 16 to 32 segments. Setting each LED individually is possible via the UDP API: https://github.com/Aircoookie/WLED/wiki/E1.31-&-UDP-Realtime-Control

trdb33, 2019-12-06:
can I use color palettes to include my personal colors ? 
what I'm looking for is just be able to use a list of colors for a static ambilight, leds are installed behind a canvas, I've used many an arduino nano years ago to use this methode with a combination with a Photoshop script that extract median border colors as you can see in the attached capture
https://www.dropbox.com/s/3664vajns861rql/IMG-456ef3367cd70700881201a3fbf41211-V.jpg?dl=0

closed 2020-04-11 by stale[bot]