wled/WLED #224: Feature Request: Make GPIO a Build Variable or Configurable via GUI

MrPaulAR, 2019-09-18:
What do you think about making the GPIO pin a variable for the LED output?  The biggest use case I've ran across is when using a ESP32 board (like the one modeled after a D1 Mini) GPIO2 on the ESP8266 is GPIO16 on the ESP32 version.  

Sure, it's easy to change in .h file but if that could be moved to other places I think that would make it less error prone.  I think the platformio.ini file may be the ideal place as it can be either defaulted globally or set in each board profile.  I don't know if that's an option but if not allow it to be set via one/some/all of the various UIs.

intermittech, 2019-09-18:
I'm running into this issue too, I'm the creator of the [QuinLED-Dig-Uno](https://quinled.info/) board which allows you to plugin a D1 Mini ESP8266 or ESP32 based board and makes setting up Addressable LED strip a bit easier and safer with reverse polarity protection, a fuse, etc.

Sadly the pinout between the two aren't 100% compatible and while GPIO2 works fine with an ESP8266 on there, using the ESP32 module on the QuinLED board doesn't work with the defualt pinout configured.

I realize you need to use GPIO2 on the ESP8266 because of hardware constraints. The ESP32 however is much faster and basically you can send out the LED Control commands on any pin and GPIO2 is actually not a commonly used one because the onboard LED is on there too. Making it configurable via the web interface/app would be a very desirable feature!

MrPaulAR, 2019-09-18:
Heh, it's your board I'm running this on :)  Highly recommended!

Aircoookie, 2019-09-18:
Alright, maybe I can add a conditional setting to change the pin at runtime for ESP32 at least!

jschweizer78, 2019-10-15:
The issue I am having is on a NodeMCU both GPIO 16 (yellow) and GPIO 2(blue) are internal LEDs. I want to use GPIO 3 (as in code comments), However, when I try to compile from src but I get errors on the build regarding files not found in ".pio/build/..". How I fix this or can LED be configured from GUI to support these different boards? NOOB to platformIO

Vendo232, 2019-10-20:
I would like to change GIO2 to GIO4 , how can I do it?

Aircoookie, 2019-10-21:
@Vendo232 change the `#define LEDPIN` in NpbWrapper.h to 4. Also please note that 4 is the default pin for IR, so you need to change IRPIN to something else as well.

closed 2021-06-30 by Aircoookie