wled/WLED #226: Add Ability to light certain LEDs Via MQTT and/or HTTP API

smileyhogue, 2019-09-21:
Would it be possible to add the ability to set the color of a certain LED(s) via MQTT or HTTP API? This is something I feel could be extremely useful for home automation folks

brentbrooks70, 2019-09-21:
This should be doable in 0.9.0 with segment support, you can look at the [JSON-API wiki](https://github.com/Aircoookie/WLED/wiki/JSON-API) for additional details. I think the idea is to allow 16 or 32 segments, possibly depending on hardware. Or you could use one of the [real-time controls](https://github.com/Aircoookie/WLED/wiki/E1.31-&-UDP-Realtime-Control) to facilitate this now with control over every LED. The number of max LEDs supported depends on the protocol selected, with DNRGB supporting the largest number by splitting control into multiple packets.

Aircoookie, 2019-10-31:
Of course segment support will be a perfect fit for this, but since the number of segments will be limited to at most 32 (for memory and performance reasons), it would be nice to have a solution for larger installations.

A potential idea would be adding support for JSON commands like:

```json
{
	"seg": [{
		"single": [
			[255, 0, 0],
			[0, 255, 0],
			[0, 0, 255],
			[255, 255, 255]
		]
	}]
}
```

amd with specifying the LED index:

```json
{
	"seg": [{
		"single": {
			"0": [255, 0, 0],
			"52": [0, 255, 120]
		}
	}]
}
```

It would be of limited use to home automation though because it would be impossible to retrieve the values again. For that, segments are better suited.

lucavinci, 2020-08-26:
I would also love this feature! Any news on this?

Aircoookie, 2020-08-26:
Recently got reminded by another user as well! Time to add this soon :)

Aircoookie, 2020-08-29:
Done! https://github.com/Aircoookie/WLED/wiki/JSON-API#per-segment-individual-led-control

closed 2020-08-29 by Aircoookie

klausmersmann, 2020-10-27:
It seems that the JSON structure hat changed meanwhile? Trying the examples on the linked API-Page i'm not able to set a single LED via curl and the JSON API. Maybe you could post a working example (with curl)?

stevenmills, 2021-03-23:
I've got this working well with the JSON API as a proof of concept.  I'm making a BINGO board, and what would be nice is if the individual segment control was extended to on/off.  Example: so seg 1 has 50 LEDs and it's currently running gradient effect-- would be awesome to turn 0 - 50 of those light on/off _in pattern_ as if they were all on.  So I can POST the equivalent of B4 (before what?), as say RED (or another effect?) as the currently call number, and the previously called numbers would be running the effect.  Anything uncalled would be off/blank.

jkwim, 2021-05-26:
> It seems that the JSON structure hat changed meanwhile? Trying the examples on the linked API-Page i'm not able to set a single LED via curl and the JSON API. Maybe you could post a working example (with curl)?

Adding a comment to an old question hoping that somebody will benefit from it and also that somebody can suggest me an explanation for what I have observed.

Following worked for me if I have the default Segment (Segment 0) defined freshly. I am using 0.12.0.

` curl -d '{"seg": [{"id": 0, "i": [[255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0]]}]}' -H "Content-Type: application/json" -X POST http://192.168.1.106/json`

Response:
`{"success":true}`

However if want to get it to work on a segment added additionally, I had to always issue a dummy command to Segment 0 first and then issue the command to modify Segment 1 and so on.

`'{"seg": [{"id":0,"i":[0,[0,0,0]]}]}'`

`'{"seg": [{"id":1,"i":[0,[0, 0, 158]]}]}'`
`'{"seg": [{"id":2,"i":[0,[0, 0, 158]]}]}'`
`'{"seg": [{"id":3,"i":[0,[0, 0, 158]]}]}'`

FelixSelter, 2023-12-26:
New docs url: https://kno.wled.ge/interfaces/json-api/#per-segment-individual-led-control

commit 36e0a1eb (2020-08-29): Individual LED control via JSON (closes #226 )