wled/WLED #228: WLED and external input device(s)

huggy-d1, 2019-09-23:
I am working on a project to illuminate a dark stairway with a smart LED strip, custom effect(s), using 2 motion sensors to trigger the LED strip in forward or reverse.
This seems like a common enough problem to trigger the LED strip based on motion or button, or some sensor wired to the same ESP.
What is already supported in WLED and what would need to be in the custom portions to do something like this?
I read through all issues, the wiki, and guides and if it's already there, sorry, but I missed it.

Aircoookie, 2019-09-24:
Hi!
Right now the only input device types supported by WLED are a simple pushbutton and infrared.
Supporting more sensors is outside the scope of the project for now as the requirements of most people are to specific for one generic implementation. (What sensor model? What should it do? When will it turn off again? What activation threshold? What pinout? and so on...)

There are two suggestions I have for you:
1. You could use a different sensor and a home automation platform or similar solution. This is more advanced and configurable, but probably overkill for your and many other's needs.

2. Code the desired behavior yourself in the `wled06_usermod.ino` file. For addressing the sensor, just doing the same as in a simple sensor example sketch should work. Interfacing to WLED is a bit more arbitrary, a few details are on this page: https://github.com/Aircoookie/WLED/wiki/Add-own-functionality . I should really improve upon and document the internal API so that you can modify the code to your needs more easily :) 
The steps you'd need to take is to read the data from the sensor and trigger a certain effect (you could use Wipe and see how long it takes to fill the strip, after that switch to solid, then turn off after 1-2 minutes)

Aircoookie, 2019-09-24:
However, simple motion sensors with a configurable timeout might be a widespread enough use case for me to offer built-in support at some point.

Mariu86, 2019-09-24:
A motion sensor will be a good addition, I use the wled for all my ledstrip lights in my house and I don't have them connected all the time to the internet, I use the Infrared remote and the button to open the lights, and a motion sensor will be a more useful trigger, this can be used for automation sistem to, if you can publish the trigger to home assistant or some other automation I think.

raoulteeuwen, 2019-09-25:
> A motion sensor will be a good addition, ... or some other automation I think.

Just wondering: wouldn't adding a motion sensor to WLED make WLED more like a home automation system? People using WLED could also tie in a/the motion sensor into a home automation solution, in which WLED is also integrated, make a rule (in the home automation solution) triggering WLED when a the motion sensor is triggered?

But if @Aircoookie is willing to add it to WLED at one point: no problem of course.

Mariu86, 2019-09-25:
I think the motion sensor will be more useful in home automation if you have more wled setups with motion sensor you could se were it's movement and start an automated scene close the door or opens the blinds start the music etc.

huggy-d1, 2019-09-26:
Even though my original request was very specific, I was thinking about many other use-cases, such as "welcome home" and under-cabinet WLED-controlled LED strips.  Sure under-cabinet addressable LEDs is not the standard go-to, but with the SYNC capability, it seems like it could take ordinary boring under-cabinet LED lighting to the next level and beyond from a WOW factor perspective.

You know, seeing the under cabinet lighting dancing in precisely the same way as other WLED strips sync'd up.

Begs the question, are there multiple SYNC channels available?  _That_ is probably in the wiki if I knew where to find it.

huggy-d1, 2019-09-26:
Even though my original request was very specific, I was thinking about many other use-cases, such as "welcome home" and under-cabinet WLED-controlled LED strips. Sure under-cabinet addressable LEDs is not the standard go-to, but with the SYNC capability, it seems like it could take ordinary boring under-cabinet LED lighting to the next level and beyond from a WOW factor perspective.

You know, seeing the under cabinet lighting dancing in precisely the same way as other WLED strips sync'd up.

Begs the question, are there multiple SYNC channels available? I see grouping WLED ESPs with different UDP sync ports is the answer to that.

Morphy99, 2019-12-04:
Has anyone got any further with integrating a PIR directly into a WLED ESP? Wouldn't this be as simple as having a GPIO pin monitor?

fishbone-git, 2019-12-04:
Simple approach, just use ESPHome with another ESP device.  it has built-in support for a number of PIR devices and you can sync it directly with WLED via MQTT. Just takes a slight bit of extra money for the second ESP.
Slightly harder would be the usermod.ino.  It's not too hard to find examples you can use for a starting point and building WLED on your own is also not too difficult. More research & time than an ESPHome build, though.

Morphy99, 2019-12-04:
Yea I did think about having a second device to control the PIR sensors, wouldn't I need a MQTT broker running to do this though?

fishbone-git, 2019-12-04:
for MQTT yes, but a pi will run mosquitto easily enough. ESPhome also has capability to easily create a gpio or relay output, which then could be connected as the "button" input to WLED.  It may also be possible to add a simple lambda into the ESPhome config that makes an API call directly to the WLED web ui.  

Personally, I just pulled the code and make my own changes.  I have some experience with coding so that helped. WLED does build somehwhat easily with a number of tools (I'm using vscode).  Just took some setup time and some edits of the platformio.ini file).  I don't think it's the preferred build environment, however, as it does mysteriously fail sometimes for no clear reason.  but, it just takes a few more clicks to try again so that's good enough for my needs.

Mariu86, 2019-12-04:
I am interested about this subject... if you have a user mode for this please post it. Thanks

MrQu33f, 2019-12-12:
I'd also be very interested in this as I want to do exactly the same thig the OP wants to

rawframe, 2019-12-24:
> Has anyone got any further with integrating a PIR directly into a WLED ESP? Wouldn't this be as simple as having a GPIO pin monitor?

Firstly a big thumbs up to Aircookie!!

**Motion sensor : hc-sr501** (5v input, 3.3v output, adjustable sensitivity and duration). 
Set the jumpers to single trigger and the running time to minimum to imitate a button press (High signal). Depending on the adjust-ability of the motion sensor you may be able to fine tune the button press down to short or long press despite the datasheet claiming 5s min. For my purposes I set the wled short and long button press to point to the same macro.
WLED button press on gpio0 (\wled00\wled09_button.ino) is set to High and the button press is triggered when gpio0 reads low as a result of the button press (wled00\wled09_button.ino). So to make the motion sensor work directly you would need the opposite and so change the code to start Low and detect button press by reading High. This is fine if you you want to change the code every time you update the software but that seems like too much effort long run. **I would hope in the future that Aircookie integrates the same button press code in reverse on another gpio pin so that any motion etc sensor can direct trigger macros via high input; avoiding the extra circuitry below**.

**So in the meantime the easiest solution was to introduce a relay in the form of a pn2222 npn transistor which costs very little. Connect the collector pin to the gpio0, emitter pin to the ground pin, and the base pin to the motion sensor control pin. This works but to avoid possible shorts and have more stable pin values add a 0.22k resistor between base pin and control pin, and a 10k resistor between gpio0 and the collector pin.**

Mariu86, 2019-12-25:
I am interested in this subject to !

Aircoookie, 2019-12-28:
@rawframe Awesome, thank you for sharing how you got it to work!
I believe I will add an option to invert the button logic level soon, which would eliminate the need for additional hardware :)
Have a great start into the new decade!

closed 2020-05-03 by stale[bot]

bodylinksystems, 2020-05-13:
I'd really like more info on this.  I already have MQTT running another project via Node-Red and am trying to trigger my LEDs using a motion sensor.

JoeRedman, 2020-05-27:
I sent you an email, but I am also interested in triggering a 'scene' in WLED, according to a GPIO input. Not new at coding, just new at arduino/esp32. I am looking to say activate a chase scene in RED when GPIO 4 is high, GREEN when 5 is high...etc. Thank you, and I love WLED and all it functionality. I wish this was built-in, but it also gives me the opportunity to learn some more coding!

v0r73x, 2021-01-01:
> 
> 
> @rawframe Awesome, thank you for sharing how you got it to work!
> I believe I will add an option to invert the button logic level soon, which would eliminate the need for additional hardware :)
> Have a great start into the new decade!

Hi @Aircoookie,

I just whnat to know if that motion trigger by push bottom is built - in since I cannot find this in the feature list - or I see the tree not in the woods ;) 

thanks and happy new year 2021!

nicedevil007, 2021-01-09:
Same here, is it already implemented?

NL-Splinter, 2021-02-05:
Ok, another one interested: Wemos D1 with hc-sr501 attached. My ideal situation would be to have a MQTT event fired when the pin is triggered. I can rework the output anyway in Node Red.

nicedevil007, 2021-02-06:
My solution for now is this from the examples here on github, that works pretty good :)

usermod.cpp
```
#include "wled.h"

//Use userVar0 and userVar1 (API calls &U0=,&U1=, uint16_t)

// PIR sensor pin
const int MOTION_PIN = 4; // D2
 // PIR MQTT topic
const char MQTT_TOPIC_PIR[] = "/pir";

int prevState = LOW;

//gets called once at boot. Do all initialization that doesn't depend on network here
void userSetup()
{
  pinMode(MOTION_PIN, INPUT);
}

//gets called every time WiFi is (re-)connected. Initialize own network interfaces here
void userConnected()
{

}

void publishMqttPIR(String state)
{
  //Check if MQTT Connected, otherwise it will crash the 8266
  if (mqtt != nullptr && millis() >= 60000){
    char subuf[38];
    strcpy(subuf, mqttDeviceTopic);
    strcat(subuf, MQTT_TOPIC_PIR);
    mqtt->publish(subuf, 0, true, state.c_str());
  }
}

//loop. You can use "if (WLED_CONNECTED)" to check for successful connection
void userLoop()
{
  if (digitalRead(MOTION_PIN) == HIGH && prevState == LOW) { // Motion detected
    publishMqttPIR("true");
    prevState = HIGH;
  } 
  if (digitalRead(MOTION_PIN) == LOW && prevState == HIGH) {  // Motion stopped
    publishMqttPIR("false");
    prevState = LOW;
  }
}
```

NL-Splinter, 2021-02-06:
Very nice, exactly what I need. On to compiling then....

solarkennedy, 2021-07-10:
I solved this in a slightly different way in case it is useful to anyone.

I used a HC-SR501 + Wemos D1 (doesn't matter too much), hooked up the output in of the motion sensor to the push button pin (D3 in my case). No other hardware.

For software, I'm using v0.13,  Button 0 (0) is set to Push Button.
Then for Button 0, long press, I have it activating Preset 1.
My preset 1 is:

```
A=255
NL=1
```

Which:
* Turns on max brightness
* Activates a sleep timer for 1 minute

This isn't the most elaborate setup, you only get 1 effect here (Although you could do more stuff with the API, or use `FX=~` to cycle through effects, etc).

Without the sleep timer, the sensor is too noisy. I experimented with `T2` to toggle  things on and off, but you can't guarantee that you will get an even number of events from the motion sensor. 

But, the end result is that there is WLED-driven light when motion is detected, and it says on for a minute after there is no motion!

PawPawGranPaw, 2022-06-16:
Late comer, worked in electronics and on computers for well right after that fruity computer company startup.  I was raised a hardware person but have been forced into the soft side.  Started out wire wrapping prototype circuits and building circuit boards from scratch.  I can remember manually programming memory addresses and using ultraviolet light to erase failed EPROM loads.  I plan to make every light in our house DC, not just LED.  I have successfully built multiple projects using the ESP8266 and ESP32 programmed with the WLED ( WLED is the bomb! Thank you so much Aircoookie ) with Buttons and IR sensors (GPIO0), and Mics (A0) with sound reactive to control single (GPIO4) and duel (GPIO1) LED strips of WS8212 and SK6812.  This issue of using one ESP8266 to receive input from a PIR is what I have been struggling with also.  Glad to see others doing something similar.  I want to connect two PIR inputs (AM312 not HC-SR501) into GPIO12 and GPIO14 for motion sensing from the bottom of stairs to the top and down a long hallway.  I have to take this to the next level by customizing WLED for this project.  Has anyone been successful at using GPIO12 and or GPIO14 to control ESP8266?  I want reserve the GPIO0 for on/off/mode selecting.

blazoncek, 2022-06-17:
> Has anyone been successful at using GPIO12 and or GPIO14 to control ESP8266? I want reserve the GPIO0 for on/off/mode selecting.

Yes, it has been done successfully using either of the two WLED usermods: _Animated Staircase_ and _Staircase Wipe_
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.