wled/WLED #229: Binaries for APA102

pashis, 2019-09-23:
Would someone be nice enough to share a compiled version for APA102 leds?

Thank you

HimbeersaftLP, 2019-09-24:
https://github.com/Aircoookie/WLED/releases

pashis, 2019-09-24:
My APA102s just emit bright white light.  Can't seem to control anything via the app. Any hints?

Aircoookie, 2019-09-24:
@pashis I'll send you a compiled binary once I get home. You use ESP8266, right? I don't personally use APA102, so I don't know why it could fail.

@HimbeersaftLP thank you, but the release binaries are for WS2812B only. APA102 need a recompile.

pashis, 2019-09-24:
Great.  Yes using a nodemcu.

Aircoookie, 2019-09-24:
@pashis Here you go: https://drive.google.com/open?id=15MDKROaJDyBirhBiLaFxUQD0hwwBJ7LL
If it works for you, I'll add it to release binaries!

pashis, 2019-09-24:
Thanks aircookie.  I'll give it a go

pashis, 2019-09-24:
In terms of 'LED Preferences' or any other settings, is there something I should note?

Aircoookie, 2019-09-25:
No, just set your LED count normally and the milliamps limit to the rating of your power supply!

closed 2019-09-27 by Aircoookie