wled/WLED #235: Ws2811

Bpasop, 2019-09-28:
Hi there, [I’ve got these ws2811 lights](https://star.aliexpress.com/share/share.htm?image=UTB8C3g0M4HEXKJk43Jeq6yeeXXap.jpg&businessType=ProductDetail&title=%25E2%2582%25AC+7%252C21++9%25EF%25BC%2585+Off+%257C+50+stks%252Fpartij+12mm+WS2811+Full+Color+LED+Pixel+Licht+Module+DC+5+V+Waterdichte+IP68+RGB+kleur+2811+IC+digitale+LED+christmas+Light&platform=AE&redirectUrl=https%253A%252F%252Fwww.aliexpress.com%252Fitem%252F32957179920%252F32957179920.html%253F%2526srcSns%253Dsns_Copy%2526tid%253Dwhite_backgroup_101%2526mb%253DnOyGgrR0xaQw4lq&tt=sns_Copy&aff_platform=default&cpt=1569676823845&sk=LSeTz6LW&aff_trace_key=8d494187949e44ba8c3dab873fc46e6c-1569676823845-02521-LSeTz6LW&terminal_id=f24296bb76c148948da0399e12988d84)  and i can’t get these work good with WLED. In the user interface i’ve selected solid green color. But the lights keep chancing color and brightness. Also all the other effect or colors don’t work the way they should. Are mine lights not compatible? Or do I have to change some settings?

Kind regards

[image]

theshadowsvk, 2019-09-28:
If you selected solid (effect) the colors should be the same, if you change colors on rgb or hue slider all the leds should be the same color. Can you share your wiring diagram, especially that led panel

Bpasop, 2019-09-28:
I’ve connected the red to the 5v power source the white on ground and the green to D4 on my wemos d1 mini. The wemos and led’s do have different power supply’s. 

[image]

theshadowsvk, 2019-09-28:
Do they share the common ground ? D4 is which gpio ??

Aircoookie, 2019-09-28:
Yes, doesn't seem like the Wemos and LEDs share common ground! Please connect them together (but not the 5V!) and it should work fine!

Bpasop, 2019-09-29:
Yeah this did the trick! Thanks for the help. Only it did not worked with separate 5v, so I controlling and feeding the led's directly from my Wemos. I don't get power issues at this moment.

closed 2020-01-07 by Aircoookie