wled/WLED #236: Make change in html

mehranviva, 2019-09-28:
Hi thank to your very good project i use it and very ecxiting to me
I have a question plz tell me how can i make change in html code and see that in compiled code?
I compress it like help in header files but dont see any change at all
Sorry for my bad english:)

Aircoookie, 2019-10-02:
Hi! Easiest option is if you set the headers in `wled00.ino`:
```cpp
//#define WLED_DISABLE_FILESYSTEM    //SPIFFS is not used by any WLED feature yet
#define WLED_ENABLE_FS_SERVING   //Enable sending html file from SPIFFS before serving progmem version
#define WLED_ENABLE_FS_EDITOR    //enable /edit page for editing SPIFFS content. Will also be disabled with OTA lock
```

Then you can go to the `[espIP]/edit` subpage and upload your own `index_mobile.htm`.

mehranviva, 2019-10-02:
Thx a lot

mungle, 2020-05-23:
Hello! It will be useful having a (maybe under password as OTA updates) toggle in the setting page where you can enable or disable the editor.
Uploading the build is very easy and without pain... I would like to add personal adjustments to the html and I’m to lazy to building WLED from scratch only for a few html lines of code! 😂😅