wled/WLED #239: is a color calibration for WS2812 possible

TechnoHoschi, 2019-09-29:
Hello,

First, thanks for the great work.

Is it possible to calibrate the colors of the WS2812?

Especially with Alexa they do not quite fit. Warm white is slightly yellowish, gold goes orange. Under Hyperion, the whole thing is not a problem, since this even offers a color adjustment.

As an amateur outsider I could not find anything suitable in the source code.

Regards

Michael

Aircoookie, 2019-10-02:
Hi Michael,

there isn't really such a feature unfortunately.
WLED does use Gamma correction which makes the colors look pretty close to what I see on the monitor. You can change the gamma8 array in the bottom of `WS2812FX_fct.h`, but apart from that, there is no easy color correction.

Regards

Christian

TechnoHoschi, 2019-10-03:
Hello Christian,

Thanks.

do you mean the "WS2812FX_fcn.cpp"?
could it be the following lines?

`  uint8_t r = (color >> 16) & 0xFF;
  uint8_t g = (color >>  8) & 0xFF;
  uint8_t b =  color        & 0xFF;`

so if I have too many red, i could reduce it to "0xE0"

Regards

Micha

TechnoHoschi, 2019-10-03:
Hi,

Changes to the "WS2812FX_fcn.cpp" were successful. Now I just need some time to set the correct colors.

Thank you very much

Micha

closed 2019-10-04 by TechnoHoschi