wled/WLED #240: super slow effects?

jaygrovr, 2019-10-01:
Does anyone know how to slow down the effects to very very slow?  I have a display in my house and i want the colors to fade over minutes, is this possible?

Thanks
Jason
PS.  excellent work on this, its amazing.  I will contribute once I figure out this issue.

Aircoookie, 2019-10-02:
Hi!

The effect aren't really supposed to run slow enough for your use case. I recommend Palette or Colorfox with a low speed though - looks amazing!

Aircoookie, 2019-10-02:
Ah sorry, i was meaning Twinklefox! Look for them in the list of effects, pretty far down. Twinklefox is the last one.

jaygrovr, 2019-10-02:
I still don’t see it, which effect number is it, and how would I set it for 5-10 minute color fades?
ThanksJason

jaygrovr, 2019-10-05:
If i dontated to your software can i get a new effect made.. super slow color fade?  Timing from 1 minute to 10 minutes transition time?
ThanksJason

Aircoookie, 2019-10-06:
I've just made changes to the random color effect (FX=5) in latest master source :)  Set the intensity to 255 to get smooth fading between colors, 128 to fade half of the time and stay on the color the other half. Use the palette list to set the colors you'd like. Just replace line 207 in WS2812FX.cpp:
```cpp
//uint32_t cycleTime = 200 + (255 - SEGMENT.speed)*50;
uint32_t cycleTime = 60000 + (255 - SEGMENT.speed)*2352;
```
it will set the fade time selectable between 1 and 10 minutes using the speed slider!
Let me know if you need a binary!

jaygrovr, 2019-10-06:
Thank you sir, what is the best way to contribute to your software, monetarily?
Jason

Aircoookie, 2019-10-06:
Hi Jason,
feature requests, bug reports and pull requests are also all great ways to contribute!
If you'd like to buy me a coffee: https://paypal.me/Aircoookie

I wish you a lot of fun WLED!

jaygrovr, 2019-11-03:
Hey there aircookie, thanks for making that change to your code for me (the super slow effect).
Wanted to know if you can maybe make another change for me.
I wanted to build some icicles and each icicle would have a certain number of pixels in them (all being the same).. Can you make an effect of dripping, and the modifier is the number of pixels per icicle (from 1-30)?
Thanks
jason

jaygrovr, 2020-03-29:
yea it worked out fine.. thank you sir

closed 2020-03-29 by Aircoookie