wled/WLED #241: Lightning effect only works on half of lights

kcoop99, 2019-10-02:
I'm running 450 WS2811 pixel nodes and all the effects work without issue except the lightning function. I have them installed on my house in anticipation for Halloween! Any idea what could be going on? I'm using a 12v 30a power supply for my lights if that makes any difference.

theshadowsvk, 2019-10-02:
There is hardcoded limitation to 300pixels when using DMA method.

kcoop99, 2019-10-02:
I'm assuming that is just for the lightning effect then? Is there a way around it? Thank you for your help!

kcoop99, 2019-10-03:
So no one thinks I was talking to myself, a user commented:
"There is hardcoded limitation to 300pixels when using DMA method."

Any input is greatly appreciated.

Aircoookie, 2019-10-03:
Thank you for the report! I can confirm that the Lightning effect was broken when using more than 255 LEDs. There were 8 bit functions used erroneously. Just fixed it in latest master source! If you need a binary, please tell me if you are running 8266 or 32.

kcoop99, 2019-10-03:
Thank you! 8266. I will need to flash the entire d1 again correct?

Aircoookie, 2019-10-03:
Alright, here you go: https://drive.google.com/open?id=1WG-yafhInvk8C5JIq3iYPSzEU03mgbbc

No, it should be super easy! Go to security settings, click "Manual OTA update" and upload the bin file!

closed 2020-01-07 by Aircoookie