wled/WLED #242: Unifi / host name

omygolly, 2019-10-02:
Hi, 

I'm having problems connecting my NodeMCU with out Unifi-network at work (it works at home). After some googlin' it seems to be a problem stemming from an unknown host name, aka set a proper host name and it will get an adress. 

Any way to set the host name in the UI or otherwise?

Cheers,
B

Aircoookie, 2019-10-02:
Hi,

sounds like your company network is an enterprise network. Those aren't easy to connect to, currently WLED only supports home networks (WPA2-PSK). It seems like it is possible though: https://www.google.com/search?q=esp8266+enterprise+wifi

I'll keep this in mind and maybe I can add support soon!
Cheers!

creativeretail, 2019-10-04:
Hi

I've had problems with UniFi in the past with non-compliant hostnames. Specifically ones that included spaces. It might be worth a try amending the 'serverDescription' from "WLED Light" to "WLED-Light".

The 'serverDescription' variable is used to set the hostname. I guess it would possibly make more sense to split this out into two separate settings. You could perhaps automatically populate the hostname field with a sanitised version of the serverDescription. Basically all that's technically allowed is a-z, 0-9 and the 'hyphen-minus' character (0x2D). Some systems will tolerate anything at all for the hostname though, even unicode! 

Steve

Aircoookie, 2019-10-04:
@creativeretail Thank you for that observation! I will change the default server description to comply with that.

creativeretail, 2019-10-04:
Not saying this will solve omygolly's issue, but certainly worth a shot!

thedannymullen, 2019-11-27:
I have UniFi and 0.86 works for me.  Must be fixed now.

githubgoucho, 2022-01-24:
Hi,

I use an AVM Fritz!Box and had the same problems -> no WLED hostname inside the client list and after every reboot a new IP-adress from the esp ...

I found my router did not eat hostnames without ending zero, 
I fixed it by adding a 0x0 at the end of hostname like

  WiFi.setHostname(hostname + 0x0);

now every think works fine always a proper hostname and the same IP

cheers GG

blazoncek, 2023-01-25:
If you are having similar issues as noted above, try to upgrade your router firmware. Then check if mDNS is correctly working on your router (use mDNS/Bonjour explorer/browser).
If you then still experience host name not being resolved re-open this issue.

closed 2023-01-25 by blazoncek

weiqinglong51, 2023-01-25:
发自139邮箱