wled/WLED #243: Cannot light the LEDs even though Arduino Mega + FastLED works fine

cyberpok, 2019-10-02:
I am trying to get WLED work for the first time (total noob). But I cannot seem to light any LEDs despite the fact that my setup works just fine with an Arduino Mega running the FastLED library.

I have two 5 meters strips of WS2812Bs (10 meters/600 leds) which I'm powering with a 5V 60A power supply.
I can run the FastLED demos using an Arduino Mega, meaning I am able to supply enough power to the strips and the data flows from Arduino to the strips without any issues. However when I switch the DIN of the leds from Arduino D7 to my ESP's D4 (GPIO2) nothing happens. Both the Arduino and the ESP GND are connected to the power supply V-. [Here is the problem demonstration I recorded.](https://www.youtube.com/watch?v=D9yQFKFbP4A&feature=youtu.be)

I compiled my ESP8266-12E as [described](https://i.imgur.com/oPEOvcC.png) and loaded up [WLED_0.8.5_ESP8266.bin](https://github.com/Aircoookie/WLED/releases/download/v0.8.5/WLED_0.8.5_ESP8266.bin). Right now I can access the UI without any problems (I can see the changes I made from my laptop in the iOS app and vice versa). Even though every now and then I get a ["Device Unreachable" message on the iOS app](https://user-images.githubusercontent.com/56058675/66055367-1edc5500-e53e-11e9-8846-269d475e9fa2.jpeg), I am always able to observe the color changes on the browser.
I set the "LED count" to "600" and "Maximum Current" to "65000" mA.

As I said I'm a total noob, what am I missing?

Aircoookie, 2019-10-02:
Hi! You did everything correctly as far as I can see!

Please try using a different breadboard wire, they are very unreliable at times. Just to make sure!
If that isn't the issue, you'll likely need a 3.3V to 5V level shifter for your LEDs. The ESPs 3.3V signal voltage is technically too low for WS2812B - your Mega runs at 5V and thus works.

cyberpok, 2019-10-04:
> Hi! You did everything correctly as far as I can see!
>
> Please try using a different breadboard wire, they are very unreliable at times. Just to make sure!
> If that isn't the issue, you'll likely need [...]

Thank you! I've been following your advice.

Firstly, I tried different wires just to be sure, no luck.
Then I grabbed [this level shifter](https://user-images.githubusercontent.com/56058675/66233999-49244300-e6f5-11e9-8f29-617f61d44de7.png) and still no luck. I read something about this level shifter not working great with LEDs so I replaced it with[ this one](https://user-images.githubusercontent.com/56058675/66234088-78d34b00-e6f5-11e9-94eb-73f26b605cf1.png) today. Still no luck. Just to be sure the level shifter is not faulty I tested it with the Arduino signal (5v) as the input and I was able to output the same signal, meaning the LEDs were able to run the FastLED demo. Maybe this is not a valid way to test the level shifter since both the input and the output are 5v :) 

I wanna do further debugging. I'm thinking about monitoring the D4 pin on the ESP8266 just to be sure that it's working. Do you have any other ideas for me?

cyberpok, 2019-10-04:
I tried this build which uses D5 that was referenced on #245 
_Originally posted by @Aircoookie in https://github.com/Aircoookie/WLED/issues/245#issuecomment-538352457_
However, I still cannot get any response from the strips. Which makes me question again, what am I doing wrong?

Aircoookie, 2019-10-04:
I'm so sorry that you are experiencing all these issues! I do not know the cause, everything you did seems correct.
There are only two things left you could check:
1. Compile the FastLED or another neopixel test sketch for your ESP8266 and see if that works.
2. Try a different ESP. Maybe it is a hardware fault.

cyberpok, 2019-10-05:
Haha I'm the one who should be sorry. Apparently my ESP wasn't powered through the 3v3. I told you I was a noob :)

Thank you so much for the amazing one and only WLED @Aircoookie, you rock!

closed 2019-10-05 by cyberpok