wled/WLED #244: Can't get WLED to show up in Home Assistant

kcoop99, 2019-10-03:
Super noob here learning this stuff on the fly. I have everything working great OUTSIDE of Home Assistant but when in HA , Developer Tools, States, I cannot see WLED. My MQQT broker shows this so I believe it has been "discovered". Thank you for your help....and patience...
[image]

Aircoookie, 2019-10-03:
Hi, first of all I don't have a HA setup myself yet, so I don't 100% know how it works.
According to another user, you might have to  add "discovery: true" in the mqtt section of the configuration.yaml file.

smartiq, 2019-10-06:
Hi; I have been having a similar issue with HA not being able to see wled entities.  I have been tracking it down in my spare time.  My issue revolves around the wled node not actually sending the config on the wire.  I _think_ it has something to do with the config packet being too large and one of the layers drops it.  I am not certain which layer is dropping it yet.

Here is some stuff you can do to see if you are seeing the same thing I am.  If you can connect to your mtqq mosquitto instance and see what is actually being posted and we can debug a little more.  A command like:
`mosquitto_sub -v -t '#' -u username -P password`
should show you stuff getting posted to your mqtt messenger.  You will also get a few initial connect notifications (these are the things that have been posted with the retain flag set).
You will need to use a valid username/password if you have authentication setup.

When I do this, I get something like:
```
wled/69c95c/g 127
wled/69c95c/c #FF9F00
wled/69c95c/v <?xml version="1.0" ?><vs><ac>127</ac><cl>255</cl><cl>159</cl><cl>0</cl><cs>0</cs><cs>0</cs><cs>0</cs><ns>0</ns><nr>1</nr><nl>0</nl><nf>1</nf><nd>60</nd><nt>0</nt><fx>0</fx><sx>75</sx><ix>128</ix><fp>0</fp><wv>-1</wv><ws>0</ws><md>1</md><cy>0</cy><ds>WLED Kitchen</ds></vs>
```

This means the wled instance is connecting to mqtt, but some of the data is missing.  In my case, I am not seeing a post to:
`homeassistant/light/WLED_b4e62d69c95c/config`
which is needed to tell HA about the wled instance (autodiscovery if-you-will).  This should get sent every time the unit starts up.

If I hookup a serial port to the wled and debug (via println, etc), I see the onMqttConnect getting called in the mqtt ino and sendHADiscoveryMQTT() gets called, but nothing "makes" it to the mqtt host.  As such HA doesn't see it.

I logged the contents of the config packet (via serial) and injected them directly into mqtt (via mosquitto_pub) and that resulted in HA learning about the device, but it forgets about it after ~2 days so that isn't a good long term solution.

Currently I am running with the following complete hack:
```
diff --git a/wled00/wled17_mqtt.ino b/wled00/wled17_mqtt.ino
index c700488..105b0f2 100644
--- a/wled00/wled17_mqtt.ino
+++ b/wled00/wled17_mqtt.ino
@@ -176,7 +178,7 @@ Send out HA MQTT Discovery message on MQTT connect (~2.4kB):
   int i = 0;
   bool isNameStart = true;
 
-  for (uint16_t j = 0; j < jmnlen; j++)
+  for (uint16_t j = 0; j < 10; j++)
   {
     if (pgm_read_byte(JSON_mode_names + j) == '\"' || j == jmnlen -1)
     {
```

Note this solution isn't recommended, but it is better than nothing.
With this code change, the packet size for the config is limited and you do not receive most of the FX values in "fx_list", so HA doesn't learn about all the awesome effects, but it does at least allow HA to turn the wled instances on and off.  In my case, making the packet smaller does allow for the config entity to show up being written to on mqtt.

I am still debugging in my spare time.  I have basically ruled out the wled code so I am digging into the async-mqtt-client code.  I am new this this whole software ecosystem so the going is slow.  Perhaps someone else here can see farther than I.

Good luck,

Barry

ghianciulu, 2019-10-06:
i've added wled into HA using this configuration inside light section (via MQTT)

`  - platform: mqtt
    name: "LedStrip"
    command_topic: "wled/all"
    brightness_command_topic: "wled/all"
    rgb_command_topic: "wled/all/col"
    rgb_command_template: "{{ '#%02x%02x%02x' | format(red, green, blue)}}"
    effect_command_topic : "wled/all/api"
    effect_list:
    - "FX=0"
    - "FX=1"
    - "FX=2"
    - "FX=3"
    - "FX=4"
    - "FX=5"
    - "FX=6"
    - "FX=7"
    - "FX=8"
    - "FX=9"
    - "FX=10"
    - "FX=11"
    - "FX=12"
    - "FX=13"
    - "FX=14"
    - "FX=15"
    - "FX=16"
    - "FX=17"
    - "FX=18"
    - "FX=19"
    - "FX=20"
    - "FX=21"
    - "FX=22"
    - "FX=23"
    - "FX=24"
    - "FX=25"
    - "FX=26"
    - "FX=27"
    - "FX=28"
    - "FX=29"
    - "FX=30"
    - "FX=31"
    - "FX=32"
    - "FX=33"
    - "FX=34"
    - "FX=35"
    - "FX=36"
    - "FX=37"
    - "FX=38"
    - "FX=39"
    - "FX=40"
`

kcoop99, 2019-10-06:
Would you mind attaching that in a format I can copy and paste into HA?

ghianciulu, 2019-10-07:
I've tryed to use the code mode to paste it. But was not correct at all. I don't know why. I can try later after work. Sorry

todfox, 2019-10-09:
I am having the same problem.

[image]

todfox, 2019-10-09:
I am having the same problem.

Philje123, 2019-10-18:
I tried adding an MQTT light manually but couldn't get it right and the on/off was inverted and setting colours resulted in it turning off.

I deleted the MQTT light entry and tried to get HA to discover it but it wouldn't.

I then removed the intergration for MQTT and re-added it and checked the box for auto discovery and it appeared straight away.

kcoop99, 2019-10-19:
Philje123, what box did you check for auto discovery? Is this on the WLED app?

Philje123, 2019-10-19:
No. When you add MQTT through the integrations menu in Home Assistant it asks for the credentials of the MQTT server and there is a tickbox or slider to turn on discovery.

kcoop99, 2019-10-20:
So when I add MQTT via the add-on store, I am not seeing this. I see the config (I have left the username and password blank). I cannot see anything when I get to this step:
"To use the Mosquitto as a broker, go to the integration page and install the configuration with one click:

Navigate in your Home Assistant frontend to Configuration -> Integrations.
Use the Add button and search for MQTT
Configure the Broker, Port, Username, Password and Submit."

This is what I see when I go to the configured integrations menu:
[image]

Philje123, 2019-10-21:
It's the step before this. So when you click to add the integration at the bottom of the window asking or the credentials there should be the toggle for auto discovery.

Have you got discovery enabled for HA?

mmatesic01, 2019-11-09:
Hi,
I am using latest WLED version 0,86, I have Mosquitto MQTT Broker in HA integrated in "HA integrations" with discovery.
I was able to auto discover to all my tasmota devices, but no WLED. I've also writen MQTT settings into my WLED unit (Wemos D1 mini) so I can see it registering in my Mosquitto MQTT log.

What did I miss to do?

todfox, 2019-11-10:
Which firmware branch did you initially upload to the D1?

mmatesic01, 2019-11-11:
> Za: Aircoookie/WLED <WLED@noreply.github.com>
> Kp: mmatesic01 <mmatesic@live.com>, Comment <comment@noreply.github.com>

V3

Poslano iz tabličnega računalnika Samsung.

-------- Izvirno sporočilo --------

todfox, 2019-11-14:
V3? Which one of the binaries below did you install? I installed the wrong one and everything worked except for the HA function. I flashed the correct version for the D1 which for me is ESP8266.bin.

WLED_0.8.6_1911081_ESP32_ledpin16.bin
WLED_0.8.6_ESP01_1m_full.bin
WLED_0.8.6_ESP01_1m_ota.bin
WLED_0.8.6_ESP01_512k.bin
WLED_0.8.6_ESP32.bin
WLED_0.8.6_ESP32_ledpin4.bin
WLED_0.8.6_ESP8266.bin
WLED_0.8.6_ESP8266_APA102.bin
WLED_0.8.6_ESP8266_ledpin3.bin
WLED_0.8.6_ESP8266_ledpin4.bin
wled_0.8.6_ESP8266_lwip1.bin

mmatesic01, 2019-11-14:
WLED_0.8.6_ESP8266.bin

Poslano iz tabličnega računalnika Samsung.

-------- Izvirno sporočilo --------

davidclancy, 2019-12-19:
I to am having the same issues with mqtt. All my mqtt device show up in HA but not wled. All setting  are correct in wled. Puzzled

CaffeinatedFunctionality, 2019-12-31:
I also can confirm this is still a problem

davidclancy, 2020-01-01:
Aircoookie, when will you be able to solve this issue.  Will it be possible to be addressed in the currant Beta release. Thanks for the hard work.

Aircoookie, 2020-01-06:
Sorry for taking so long on this...
Since I can't reproduce these issues on my testing setup, it is very hard to find a fix!
But I'll stay on it!

closed 2020-05-13 by stale[bot]