wled/WLED #245: Using level shifter and need to change output pin

kcoop99, 2019-10-03:
I am looking to change my output pin from d4 to D5-D8 for use with my logic level shifter board/kit I purchased. Thank you.

Aircoookie, 2019-10-03:
Hi, you can do that in the file `NpbWrapper.h`. For D5, use `#define LEDPIN 14`

kcoop99, 2019-10-04:
You have to hold my hand here......I can open NpbWrapper.h on the github
page but how to do open it on the D1 mini? When I originally flashed the D1
I used the ESP Home Flasher tool and didn't use Arduino.ino. Again, thank
you for working with this NOOB!

Aircoookie, 2019-10-04:
Oh sorry, I was assuming you were compiling from source. Here is a binary for LEDPIN D5!
https://drive.google.com/open?id=1OVIoLloVQ8QhdeymhfTHF2CAurvyIc86

Just install it by clicking "Manual OTA update" in security settings. It may not be the most stable because of bitbang pin driving requirement (who designed this levelshifter that only works with D5 to D8 haha).

kcoop99, 2019-10-04:
This guy. He's been super helpful he gave me  this response:
https://www.evilgeniuslabs.org/mini-wifi-led-controller

"I didn't know about WLED, I'll have to check that out.  It does look like
they use pin 4 by default.  Pin D4 is not connected through the level
shifter on any of my Wemos D1 Mini shields, including even my 6 output
version
<https://www.evilgeniuslabs.org/hex-wemos-d1-mini-wifi-led-controller>.
I've avoided using it when possible because it has a 10k pull-up resistor
and the built in LED attached to it.  It's interesting that they chose that
as the default pin."

debsahu, 2019-10-04:
Its hardware limitation for NeoPixelBus implementation for esp8266: https://github.com/Makuna/NeoPixelBus/wiki/ESP8266-NeoMethods

vexter0944, 2019-12-05:
Thanks for the tip on d14 - re-compiled from source last night as the file from google drive above was not stable for me.  Worked perfectly with level shifter after the upload.  Appreciate it!

kcoop99, 2019-12-05:
Can you please share how this can be implemented in my current version 8.6? Can it be updated OTA? Is there a simple process? Thank you all for your work on this!

vexter0944, 2019-12-06:
I have the bin file I compiled for D5 if you like I could maybe email it to
you.  If not - the only way I know how to do this was to download the
source code for 0.86, change the npbwrapper.h file as described in the
thread and then compile and upload the new bin made from those changes.
Fair warning - I had to download several libraries (they were easy to find)
to compile the code.  But afterwards, it worked perfectly!

vexter0944, 2019-12-20:
FYI - tried same exact thing with 0.9.0-b1 - lights wouldn't even come on.  changed just the ledpin to 14 just like before - compiled (has to add new library - ESPAsyncUDP) - and nada.  Going back to 0.86

closed 2019-12-21 by Aircoookie