wled/WLED #246: Set custom palette via HTTP/MQTT

jwingefeld, 2019-10-07:
Hi,

i think this qualifies as feature request. I think it would be really convenient to be able to set custom palettes via http (maybe as json?). This could be one dedicated "slot" with a limited number of colors so RAM usage would stay predictable.

Possible use cases would be:
-  a color palette editor in the web interface
- pywal (generates color palettes for your Terminal from the current wallpaper) integration. So you'd load up a picture of a sunrise and BOOM, matching color palette on your desk.
- integration into $GENERIC_RGB_GAMING_TOOL to get colors matching your 1337-gaming-battlestation

Aircoookie, 2019-10-08:
Hi,

thank you for the feature request! I've already planned such a feature via a JSON API. Current timeline for implementation is likely December.

NietGiftig, 2020-06-09:
@Aircoookie 
Until yesterday I used my own proggy for my floorlamp leds, using node-red as home automation.
Now I have resoldered this floorlamp to the default WLED data-pin and use WLED.
Verry (very!!!) satisfied.
But this request is a very good addition to include some of my old palettes with an easy method.
I can store in, and send any posible palettes from node-red, so i hope not a huge impact to the progmem in WLED.
A limited count of free useable palette memory would be sufficient 
Any timeline possible ?

blazoncek, 2022-07-29:
It may not be JSON/HTTP or MQTT compatible but I have implemented (up to 10) custom palettes that can be stored on embedded filesystem and selected at runtime in the same way as built-in palettes.
It will still need thorough testing though as the option is part of the larger change I created (2D support and audio reactive as a usermod).
Commit: https://github.com/blazoncek/WLED/commit/df534d30bf5087888f0df644ffb8f7c5225eb263

Orace, 2022-09-07:
I'm currently working on a [palette editor UI](https://codepen.io/Oracetonick/pen/qBYdwaN).
I plan to integrate it with wled.
Feedback are welcome (I'm more of a C# guy than vanilla JS).

Aircoookie, 2022-09-07:
@Orace that is super cool and exactly what I had in mind of implementing!
A "Remove color stop" button and maybe a row of recently used colors are the only things I would consider still missing,

Orace, 2022-09-08:
@blazoncek, I plan to integrate it in WLED current UI as a _standard_ palette:

- [ ] integration
- [ ] switch to edit mode with a collapse button
- [ ] use the current color selection UI for _color stops_ (a.k.a. thumbs) 
- [x] manage thumbs collisions
- [x] manage up to 16+2 thumbs to match WLED capabilities
- [x] button to delete current thumb
- [x] button to distribute evenly the thumbs
- [ ] draw a line between the thumb and the actual matching position on the palette (UX in case of a collisions)
- [ ] tests on different devices
- [ ] dynamic palette in WLED
- [ ] API to R/W the dynamic palette

blazoncek, 2022-09-08:
I do not know if you noticed but I added up to 10 custom palettes (palette0.json, palette1.json, ...) to 0.14
You can upload palette from JS using /upload endpoint. You can as well download them into UI with the name on the URL.

Orace, 2022-09-08:
I didn't try it, and have only used the 0.13.3 release
Does the custom palette are saved in presets?
I was thinking of a per segment custom palette that will be saved in presets, like it's done with the 3 configurable colors: `"col":[[0,0,255],[255,255,255],[255,0,0]]`

Something like `"pal":-1,"grd":[[0,#ABCDEF],...,[255,#123456]]`

blazoncek, 2022-09-08:
[image]
They are part of regular palette options, stored on the file system and loaded at boot.

Orace, 2022-09-08:
As I understand it, if you use one of these custom palettes in a preset and then change the palette, the preset will use the new colors 🤔
I don't find this solution very flexible.
My proposal is to store a copy of the palette in each segment of a preset.
The only problem being the memory footprint: ~256 bytes in the worst cases / segments.
I find it acceptable in my case (ESP32 with 1500kB of file system, 14 segments, max 10 presets).

blazoncek, 2022-09-09:
It may not be *flexible* as in change it on-the-fly to whatever you want, but having 10 user modifiable palettes would solve many requests received so far.
As for truly on-the-fly modifyable palette I would rather recommend writing a usermod for that.

Your idea of storing palette within (segment of a) preset has other issues as well apart from (you think) negligible RAM footprint. This is due to the way how palettes are handled when you have more than one segment or when a transition between one palette to the other happen.

Parsing 16 steps palette needs at least 2k JSON buffer. And is slow. Remember that each segment can have its own selected palette and there can be up to 32 segments. You would need to switch that for each frame of each segment.

blazoncek, 2022-09-09:
With 0.14 you can write a custom effect in a usermod that can handle the palette wrapping on its own (this is why you see purple at the end).
You can enable or disable palette wrapping in settings, but not every effect upholds that.

Orace, 2022-11-07:
> Parsing 16 steps palette needs at least 2k JSON buffer.

🤔 I think many people are able to do much better knowing that a flat json array of integers is expected and the result should be stored in a byte array. Sticking to ArduinoJson in this case is probably a shot in the foot.

Anyway I made some progress and added the lines between the thumbs and the actual matching positions on the palette:
[image]

The [codepen link](https://codepen.io/Oracetonick/pen/qBYdwaN).

blazoncek, 2023-01-25:
@Orace it has been some time since you posted last. Would you care to add ability to upload custom created palettes using your tool to WLED file system?
Something similar that was done recently for displaying pixel art on WLED matrix (in folder /data/pixart or accessible on http://wled.local/pixart.htm)

werkstrom, 2023-04-08:
[image]

blazoncek, 2023-05-27:
Closing as custom palettes have been implemented natively in WLED UI.

closed 2023-05-27 by blazoncek