wled/WLED #247: RGB cube animations

Ga1j1n, 2019-10-08:
I'm using WLed in a 3 panel 8x8 (64led) WS2812 cube and it's awesome, however as I they are effectively low Res displays I can't help think it would be awesome be able to run animations suited to the panels as well as the string led effects, if it's feasible would you mind considering it for the future?

brentbrooks70, 2019-10-08:
The Neopixelbus library supports panels so I would imagine it would just require you to change the way the LEDs are defined originally. Here is the [wiki with info](https://github.com/Makuna/NeoPixelBus/wiki/Matrix-Panels-Support )

Aircoookie, 2019-10-08:
I'll consider adding 2D and 3D effects in the future!

Ga1j1n, 2019-10-09:
Thanks for that, most appreciated, really great projects

I noticed WLEDsquared amongst your projects, looks to me like you already
considered panels, very interesting, I've ordered a 64x32 matrix and Brian
Loughs board from Tindie, don't suppose that's anywhere near a binary
release?

regards

Dave R

Aircoookie, 2019-10-18:
Hi!
I'm really sorry to say this, but WLEDsquared was an experimental version for panels that is on hold for now because I am focusing on standard WLED and other projects. For now it would be unmanagable for me to maintain it alongside WLED.
I will very likely make updates to WLEDsquared again at some point, but not in the near future sadly (but there is a high chance after I get WLED out of beta)!

The-Techy, 2021-12-02:
Any update on this? I've been thinking about building an led cube, and would love to use WLED (Already used it in other projects, It's awesome!)

blazoncek, 2022-10-10:
0.14 will add 2D support. You will be able to set up segment for each face of the cube and have different effects play on each segment.

closed 2022-10-10 by blazoncek