wled/WLED #248: Stuttering crossfades after update to 0.8.5

StefanRu1, 2019-10-08:
Hi,

after i updated from 0.8.4 to 0.8.5 fades in brightness are stuttering.
E.g. when i turn on the lamp brightness should fade in. After the update this is only a fed with 3 steps. 
Effects are working fine.

I have the following transition settings
Crossfade: On
Transition Time: 1000 ms
Enable transition for secondary color: On
Enable Palette transitions: On

I tried here which effect it is and it is the crossfade which is doing this very low sampling / stuttering.

Any Idea?

Thanks and best regards,
Stefan

Aircoookie, 2019-10-08:
Hi Stefan,

thank you for the report! This issue should be fixed in the master source already!
You can either download that or flash this binary: https://drive.google.com/open?id=1WG-yafhInvk8C5JIq3iYPSzEU03mgbbc
Enabling "Turn on after boot" and rebooting will also workaround the issue.
I hope it will fix it for you!

StefanRu1, 2019-10-08:
Wow, thanks for the quick help.
I flashed your binary and it works like a charm.
I have also the feeling that 0.8.5 also solved my MQTT issues.
At least it seems a lot more stable now.

Thanks again for your great software and support!

P.S.: One remark the text in the browser tab still says WLED 0.8.4, but the update page say 0.8.5.

Thanks and best regards,
Stefan

StefanRu1, 2019-10-10:
I can close this, everything runs smooth and stable.
Thanks!

closed 2019-10-10 by StefanRu1